wled/WLED #3295: 2D Matrix Setup Not Saving After 19 panels.

bigsandman69, 2023-07-20:
### What happened?

I recently updated to the newest version of WLED beta release 0.14.0-b3. I am trying to use the 2D matrix setup for 24 8x32 led matrix panels. It will allow me to setup the 24 screens but when I hit save and go back into 2D configuration it will not save any past 19.

### To Reproduce Bug

Add 24 matrix panels configure and save. Then go back into 2D configuration and will only show 19 panels configured and number 20 will be default and shows none past 20 in config windows at bottom.

### Expected Behavior

2D configuration will not save any matrix panels past 19.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED version 0.14.0-b3 (build 2306210)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2023-07-27:
Are your panels on one LED pin or split across multiple? You are trying to drive 6.144 LEDs which is quite a huge amount even for an ESP32. Maximum LEDs per bus (pin) currently defaults to 2048, so you'll need to split your matrix across three pins.
Are you already able to drive all LEDs in 1D mode and just the 2D setup is giving you trouble?

bigsandman69, 2023-07-28:
> Are your panels on one LED pin or split across multiple? You are trying to drive 6.144 LEDs which is quite a huge amount even for an ESP32. Maximum LEDs per bus (pin) currently defaults to 2048, so [...]

Sorry I didn't put my hardware in description. I'm currently using Wasatch Advanced 8 channel controller board & Each channel controls 768 Leds of the overall panel. I updated it from WLED 0.13.3 to use the 2D configuration to give me options to display text when my show isn't running. Currently The board drives Matrix fine from XLights & It actually drives fine in WLED it just doesn't save the configuration after panel 19 in the 2D configuration page. when you go back into 2D configuration panel 19 is in default position and no panels past 19 show unless you make a change to number of panels and re-setup every panel past 19 again, but every time you leave the 2D configuration page it goes back to 19. Is there some sort of MAX set somewhere that would cause this not to save past 20 panels total?

blazoncek, 2023-07-28:
Please post your `cfg.json` after saving 2D configuration.
You can obtain it from security settings page.

dosipod, 2023-07-29:
Seems to be pushing it , the 2D page even got reset when we added 24 panels but segments still showed the matrix  . Prior to the reset as the OP mentioned you will only be able to save until panel 19 .Sorry to butt it 
[image]
[2D_RESET_AFTER_ADDING_24_PANELS.cfg.json.txt](https://github.com/Aircoookie/WLED/files/12207407/2D_RESET_AFTER_ADDING_24_PANELS.cfg.json.txt)

dosipod, 2023-07-29:
A work around for the 2D page might be to just setup the whole thing as one matrix 
[image]

However peek is broken and i am sure real leds will also not work correctly as I think we seen issue with even less leds  here in   discord https://discord.com/channels/473448917040758787/801531333067014164/1128562990338285628 , it was unfortunately  not solved 

[onepanel_cfg.json.txt](https://github.com/Aircoookie/WLED/files/12207469/onepanel_cfg.json.txt)

blazoncek, 2023-07-30:
> 2D_RESET_AFTER_ADDING_24_PANELS.cfg.json.txt

cfg.json looks ok and has 24 panels, 8 LED outputs 768 pixels each. I would guess RAM gets depleted somewhere.
Please use a debug build and provide debug output to narrow down this issue.

> However peek is broken

Peek will not work correctly with anything above 4096 pixels. It is not designed for that.

dosipod, 2023-07-30:
Not sure my debug is helpful as I have no leds  or if the setup with one panel is working for the OP . Still using debug build i could not see anything unusual  on serial and i could add more panels but trouble start when the offset can no longer be changed   . It might help if the last panels after 19 is added one by one ( meaning add one and save ) 
[image]
[image]

blazoncek, 2023-07-30:
I think the problem is that URL becomes too long. There is nothing we can do about it unless we reprogram web server.
The limit will be 21 panels in this case.

blazoncek, 2023-08-09:
I am not sure we will be able to overcome this problem.
I will assign a backburner status and we'll see what we can do about it in the future.

dosipod, 2024-12-16:
I do not have that many to test but at least the limit in ui can be changed WLED_MAX_PANELS  , strange it is in FX.h but someone in discord was trying that  
[image]
[image]

blazoncek, 2024-12-16:
@dosipod unfortunately this change alone will not allow UI to behave correctly.
It will allow, however, to modify `cfg.json` manually and for such change to take effect on reboot.

dosipod, 2024-12-16:
@blazoncek It is unfortunate that the person with the panels no longer interested in the code change to test but in my test unit with only  WLED_MAX_PANELS  changed I could still see the 2D page with 28 panels after the reboot . 
I will see if someone has that many to do the full test with fixtures else ( which I hoped to avoid ) I will try to setup something with DDP . 

Editing cfg.json which I hope is not needed is very dirty with that many panels  as you would need  to adjust the panels x and y position  which I tried  to do before we know about the WLED_MAX_PANELS in FX.h  ,

blazoncek, 2024-12-16:
@dosipod this issue was opened for this particular reason - not able to save panels beyond 19. Since then I intentionally limited panel number in code and UI to 18 (as 19 was unreliable). Previous value was 64.
The problem is in AWS handling of large POST request. Unrelated to WLED.
If changes made by @willmmiles improved AWS handling of large POST requests then the limit may be safely increased. How much, that will depend on buffer size.

closed 2025-04-16 by blazoncek