wled/WLED #3297: WLED gives very poor performance on artnet ethernet WT32-ETH01

jagspaul, 2023-07-21:
### What happened?

WLED gives very poor performance on artnet ethernet WT32-ETH01 board.

WLED support 20 universe each universe has 170 RGB pixel data for Ethernet firmware.
I set 8 LED data each 510 RGB LEDs.
When I give full load of 20 universe over wifi and found working perfect. But same configuration when applied on Ethernet instead wifi I found very poor performance . many missing universes. with Ethernet it works perfect up-to 15 universe.

Ethernet performance is very poor then expected, even poor than wifi.

In this connection you are requested to look into the matter and find a solution for the same.

Thanks & Regards
Jags  

### To Reproduce Bug

I set 8 LED data each 510 RGB LEDs.
When I give full load of 20 universe over wifi and found working perfect. But same configuration when applied on Ethernet instead wifi I found very poor performance . many missing universes. with Ethernet it works perfect up-to 15 universe.

### Expected Behavior

Ethernet performance is very poor then expected, even poor than wifi.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.13.1_ESP32_Ethernet.bin

### Which microcontroller/board are you seeing the problem on?

ESP32, Other

### Relevant log/trace output

```shell
WT32-ETH01
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-07-22:
Perhaps a question for Ethernet driver developer.
WLED is using stock Ethernet drivers from Arduino/IDF ESP platform.

If you think it is still a WLED issue, please add some timing debug printouts and point us where the delays happen.

mxklb, 2023-08-14:
Could be related to #3239

@jagspaul do you recognize poor performance for effects/channels other than "Solid Color" and "Brightness"?

jagspaul, 2024-04-20:
I recognize poor performance on the point of speed.

jagspaul, 2024-08-18:
What developers are doing ?

Why they kept themselves silent in this matter ?

DahliaWitt, 2024-10-13:
I am also experiencing this issue on WLED 0.14.4. We are running 1,728 LEDs via ArtNet from Resolume. Works perfectly over wifi but when using ethernet, experiences horrible stuttering/flashing. The interface over ethernet is also painfully slow while WAI on wifi. Have replicated this issue across two different boards.

troyhacks, 2024-10-22:
@jagspaul @DahliaWitt 

Art-Net/network pixel data into WLED is a bit harder because we're not really optimized for listening to a lot of packets. That said, if it works currently on WiFi it should be possible to get parity with Ethernet. 

One thing you can try right now is to get WLED working with Ethernet and then remove your WiFi credentials so it doesn't connect. You may also want to disable AP mode entirety. **_Don't do this on a WLED setup you don't want to accidentally break. Back up everything before you try!_**

All my commercial Art-Net hardware uses FPGAs internally to process everything and keep it clocked perfectly  - something FPGAs are extremely good at, but microcontrollers are not.

That said, we're using an older version of ForkInEye's E1.31 code - his newer version has a ringbuffer that might help here. 

It's on my list to look into.

troyhacks, 2024-10-22:
If you're looking for something ESP32-based to render Art-Net data, here's something that's better at it than WLED:

https://github.com/hpwit/artnetesp32v2 ( and https://github.com/hpwit/artnetESP32 )

As you can (hopefully) tell, this is MUCH more bare-metal than WLED. No fancy web interfaces, just raw "network to pixels" performance and you have to setup and compile everything yourself.