wled/WLED #3307: Sinric Pro support for Google home and  alexa support

hplvm, 2023-08-05:
**Is your feature request related to a problem? Please describe.**
I upgraded my WS2812B setup from Custom DIY Sinric Pro + WS2812BFX firmware to WLED, first thing I found was there is a lot of customization here. But I could not find a way to connect to google home or Alexa. 

**Describe the solution you'd like**
Sinric Pro is a free service to connect to Google home and Alexa, I want support for adding Sinric Pro credentials in config and its functionalities implemented in the firmware. 

**Describe alternatives you've considered**
I couldn't find a already existing feature of WLED. Correct me if I'm wrong, I did go through the repo.

**Additional context**
Website : [sinric.pro](https://sinric.pro/)
SDK : [ESP32_ESP8266_SDK](https://github.com/sinricpro/esp8266-esp32-sdk)

dosipod, 2023-08-05:
Just FYI ,Alexa is supported  with wled . Not google home though

drbdani, 2023-09-14:
I would be also interested in a Google Assistant support, would be great!