wled/WLED #3312: Syntax error at overwriting presets

reald, 2023-08-09:
### What happened?

When updating a preset "SyntaxError: Expected ',' or '}' after property value in JSON at position 2226" appears.

### To Reproduce Bug

A preset "Fire 2012" exists at position 3. After some modifications trieded to update the preset with following settings:

[x] Overwrite with state
[x] Include brightness
[x] Save segment bounds
[ ] Checked segments only
Save to ID 3

### Expected Behavior

Preset is updated

### Install Method

Self-Compiled

### What version of WLED?

0.14.0-b3 "Hoshi"

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2023-08-11:
Factory reset the unit or flash the nightly builds , the error you see there could happen even with other effects  but  I can save Fire 2012 on my test unit just fine . You seeing that error with this effect is just a coincidence unless your Self-Compiled is  messing things  . I have seen this error more with ESP8266 builds when making a lot of presets but cant say what is the reason for it 
[image]

softhack007, 2023-09-17:
@blazoncek any idea on this one? 

Same as @dosipod, I have saved lots of presets with default options, however usually doing it on esp32 as I only have one 8266, which is reserved for collecting dust 😉 on my high cupboard. Could there be something causing errors on 8266?

blazoncek, 2023-09-18:
This is related to AsyncWebServer buffer being corrupt. Refreshing UI will often resolve the problem.
My speculation is that at some point buffer pointer is being released too soon in async call.
Or that send buffer gets overwritten by incoming traffic.

blazoncek, 2024-01-06:
Possibly related to #3641

closed 2024-01-09 by blazoncek