wled/WLED #3313: Advanced selection of segments

stefankoehne, 2023-08-11:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
I'm currently experimenting a lot with segments and find the option of selecting via the checkboxes a bit complicated.

I come from event technology and know from light consoles the possibility of selecting and saving individual fixture groups and using "Set", "Next", and other helpful commands.

Similarly, I would like WLED to have a few buttons above the segment area that allow the following:
"Select All" - As the name suggests, select the checkboxes of all segments
"Toggle Selection" - Invert the current selection
"Next" / "Previous" - Deselect the currently selected segment and directly select the next/previous one. If several are selected, keep the pattern (e.g. 1, 4, 7 are selected > Next > 2, 5, 8 is selected)

Another nice to have would be a "highlight" button that would temporarily light up the selected segments in white (or use a preset defined in the settings) to better identify the selection when using a lot of segments. I hope you understand what I mean.

I would be happy if someone else could be enthusiastic about the functions and the idea and maybe implement it in the future.
Best regards!

stefankoehne, 2023-08-11:
@

> Segment groups/sets are available in 0.14-b3.

I do not mean groups of pixels inside one single segment. I think about groups of segments to select multiple segments without clicking on every checkbox.

blazoncek, 2023-08-11:
exactly that. I did not say *grouping*

stefankoehne, 2023-08-11:
Okay, I found it. Sorry for the missunderstanding. That is not as comfortable as I would have imagined, but makes it a little easier.
The highlight function would still be great.