wled/WLED #3314: 0.14.0-b4 configure segments trough MQTT/JSON API no longer working

oxivanisher, 2023-08-12:
### What happened?

Up to 13.3, segments could be configured trough the JSON API over MQTT. It seems that it now only accepts the first segment and no additional anymore.
Since the MQTT-Length is/(was?) limited, I had to split up the the segment config into one message per segment.

### To Reproduce Bug

Configure multiple segments with multiple JSON messages trough MQTT.

### Expected Behavior

No existing segments should be created, existing ones should be upgraded.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b4

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-08-12:
MQTT handles larger packets now.
It also uses the same JSON processing as HTTP request does.

blazoncek, 2023-08-12:
`cln` is not used by WLED.

oxivanisher, 2023-08-12:
> MQTT handles larger packets now. It also uses the same JSON processing as HTTP request does.

Good to know, but the original problem still exists. Even if I send all segment configs at once, it is not working.

> cln is not used by WLED.

Also good to know, but if it is listed in the example, it should at least be commented with "not used" or something like that. :)

blazoncek, 2023-08-12:
> Good to know, but the original problem still exists. Even if I send all segment configs at once, it is not working.

Post your sample.

> > cln is not used by WLED.
> 
> Also good to know, but if it is listed in the example, it should at least be commented with "not used" or something like that. :)

You are welcome to update the documentation. Any help is welcome.

oxivanisher, 2023-09-10:
This problem also (still) exists in 0.14.0-b5.

> Post your sample.

Taken directly from MQTT (`wled/device-name/api` topic):
Split config variant (tested also with seconds long wait cycles between the messages):
```json
{"seg":[{"n":"tv-light-left","start":44,"len":14,"rev":true,"id":2}]}
{"seg":[{"n":"tv-light-top","start":14,"len":30,"rev":true,"id":1}]}
{"seg":[{"n":"tv-light-right","start":0,"len":14,"rev":true,"id":0}]}
```
All in one variant:
```json
{"seg":[{"start":44,"len":14,"rev":true,"id":2},{"start":14,"len":30,"rev":true,"id":1},{"start":0,"len":14,"rev":true,"id":0}]}
```
Both variants work with the 0.13.X versions and not with the 0.14.X.

Maybe this is related to #3344 ?

blazoncek, 2023-09-10:
Please use "start" and "stop" instead of "len". "len" will not work on new (to be created) segments.

oxivanisher, 2023-09-15:
> Please use "start" and "stop" instead of "len". "len" will not work on new (to be created) segments.

This did the trick! Thank you 👍

blazoncek, 2023-09-27:
@oxivanisher can you close or do you still consider this a bug to be resolved?

oxivanisher, 2023-09-27:
@blazoncek I consider it closed since it's working for me. But I can not document things I don't know about. Should we keep this open until it is documented or not?

> You are welcome to update the documentation. Any help is welcome.

blazoncek, 2023-09-27:
Just fork WLED-Docs add text into relevant file and make a PR.
You can also ask @wled-faq or @dosipod to help.

closed 2023-09-27 by blazoncek