wled/WLED #3315: Playlist toggle functionality doesn't work

ReiberAndras, 2023-08-12:
### What happened?

Toggle functionality doesn't work correctly in playlists. 
I created a playlist, which is triggered by a button press. The playlist consists of two presets: one changes the LEDs to white (no T or "on" is defined), the other turns them off after 10 minutes like so:
``{"playlist":{"ps":[1],"dur":[6000],"transition":[0],"repeat":1,"end":2,"r":false},"on":"t"}``

Unfortunately it is not possible to turn the lights off. The playlist "on" node gets overwritten from "t" to true, hence every button press will only turn on the lights.  This used to work in version 0.13.3

### To Reproduce Bug

1. Create a preset with the following API call and save it to preset id 1:
`W=255&R=0&G=0&B=0&FX=0`

2. Create a preset with the following API call and save it to preset id 2:
'T=0'

3. Create a playlist with the API command option with "on" set to toggle mode:
`{"playlist":{"ps":[1],"dur":[6000],"transition":[0],"repeat":1,"end":2,"r":false},"on":"t"}`

4. Save the playlist. The playlist executes correctly respecting the toggle. (If the light was on it turn off and vica versa)

5. Click on the playlist. The toggle functionality is broken. The light will only turn on, but never turn off. 

6. Open the Playlist:
The API command changed to:
`{"playlist":{"ps":[1],"dur":[6000],"transition":[0],"repeat":1,"end":2,"r":false},"on":true}`

### Expected Behavior

Upon pressing the button the lights turn on if they were off and turns themselves off after 10 minutes if the button was not pressed in between.
Pressing the button again turns off the lights if they were already on.

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.0-b4 (build 2308110)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-08-12:
Use a preset that turns WLED off as an end preset.
Playlist _on_ toggle works only at start of the playlist, not end.

ReiberAndras, 2023-08-12:
Thanks for the prompt response. Editing playlists directly with  /edit worked.

blazoncek, 2023-08-14:
> Thanks for the prompt response. Editing playlists directly with /edit worked.

IMO you are doing it wrong but if it works for you then all is good. 😁

closed 2023-09-28 by blazoncek