wled/WLED #3316: Turning on fan/ changing fan speeds makes some lights glitch

Sulaiman09, 2023-08-13:
### What happened?

While trying to change my ceiling fan speed I see that at the beginning and end of my led strip glitches (random colors/ some  leds turn off). sometimes it’s in other places but mainly front and back of the strip. Is this some sort of power issue? I attached a video, the color of the lights doesn’t matter
sk6812 1040 leds minus the first one. I have power injection and an 80a psu 

### To Reproduce Bug

Changing or turning on fan 

### Expected Behavior

Nothing 

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.3 (build 2208222)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

https://github.com/Aircoookie/WLED/assets/98357314/f20d0b24-e523-4d95-b39f-3f148370824e

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-08-13:
That is EMI. Use shielded data cable.

Sulaiman09, 2023-08-13:
Thank you, but could you show me what the cable is. All I'm finding are Ethernet cables and cables with multiple wires. Sorry I'm new to this stuff😅.

blazoncek, 2023-08-13:
Please use WLED [forum](https://wled.discourse.group) or Discord for hep and support questions.
Closing.

closed 2023-08-13 by blazoncek