wled/WLED #3321: Effects tab is not viewing properly on iPhone WLED App

shane-peiris, 2023-08-14:
### What happened?

When you navigate to the effects tab in the iPhone (Checked on iPhone 12 Pro Max and iPhone 14 Pro), it is all over and unable to click on any of the effects.

[image]
[image]

### To Reproduce Bug

Click on the Effects Tab. This issue is available on both the sound reactive and normal version.

### Expected Behavior

To show the effects properly just like in the web browser and without them suddenly disappearing.

[image]

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.3 (build 2210301)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-08-15:
Please open an issue in WLED app repository?
Closing.

closed 2023-08-15 by blazoncek

softhack007, 2023-08-15:
The mobile app has it's own repository on github https://github.com/Aircoookie/WLED-App

This iOS problem is already reported multiple times:
https://github.com/Aircoookie/WLED-App/issues/36 and 
https://github.com/Aircoookie/WLED-App/issues/35 and 
https://github.com/Aircoookie/WLED-App/issues/34 and 
https://github.com/Aircoookie/WLED-App/issues/39

Please, study existing report before reporting the same thing again.