wled/WLED #3322: Scrolling text direction enhancment

dosipod, 2023-08-15:
At the moment Scrolling text direction is from right to left , this feature request is to add correct movement from left to right and also vertical movement .
As you see in the video if  reverse is selected in segment then movement will change from  left to right but the fonts is  reversed

[image]

Also vertical movement at the moment is as listed in the video ( I assume the request for vertical  is for small text that does not scroll )

[image]

I have kept Feature request similar  to the original post in discord found here  ( i am only posting on their behave )  
https://discord.com/channels/473448917040758787/473448917543944193/1140772522867052596

Havenview, 2023-08-15:
For clarification I envisaged the vertical movement up and down to be on horizontal text .
eg with text such as
vvv Way Out vvv
or
^^^ Way Out ^^^

Appreciate this relies on the text length to be smaller than the display length

blazoncek, 2023-08-15:
How do you propose this to be available from UI?
I already replaced leading 0 checkmark with scrolling direction checkmark and used 3rd custom slider for letter rotation.

Havenview, 2023-08-15:
You could have the speed slider have a midpoint eg would range -128 to +127 with +ve being to the left and -ve to the right
is it possible  to use a second slider for up down and altering one axis resets the other to zero?

Havenview, 2023-08-16:
I should have used the term "another slider" instead of "second slider"

Currently:
[image]

As a suggestion add a 5th slider for the y speed?

blazoncek, 2023-08-16:
5th slider is used for character rotation so you can create vertically scrolling text.

Havenview, 2023-08-16:
Assume that's in a beta version?
I was using segment transpose for vertical scroll

dosipod, 2023-08-19:
[image]

commit 4911a74c (2023-08-16): Scrolling text enhancement.
- breaking change
- remove leading 0 checkmark
- add reverse scroll checkmark
- add vertical scroll if text fits into segment (intensity ==0 or ==255)
- rotated characters
- leading 0 check added to short texts (i.e. #DDMM0)
Fixes #3322
closed 2023-09-05 by softhack007