wled/WLED #3323: Only sparkles shows in 14.0-b4 Glitter effect, no colors.

clayclai, 2023-08-15:
### What happened?

I did an OTA upgrade from 14.0-b1 to 14.0-b4 on an ESP8266  attached to a string of 550 WS8212B. After this various glitter presets & a glitter playlist would intermittently show only the sparkles, with no other pixels being lite or bright white for all pixels. I also got this same bad result using the Glitter effect directly, without using a preset. Proper operation was restored by downgrading via OTA back to 14.0-b1

### To Reproduce Bug

OTA upgrade from 14.0-b1 to 14.0-b4 on an ESP8266. Did upgrade and downgrade again to get build # and saw same effect- had to downgrade to 14.0-b1 for proper operation.

### Expected Behavior

Expect Glitter effect & presets to operate as before upgrade

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0-b4 (build 2308110) and WLED 0.14.0-b1 (build 2212222)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

I also notice that the IP flips around from 4.3.2.1 when I started an after upgrading to 14.0-b4, up then switched to 192.168.4.1 after down grading to 14.0-b1, and back to 4.3.2.1 after upgrading to 14.0-b4 after upgrading again to reconfirm.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-08-16:
The Glitter effect (and Solid Glitter) received an update. Please experiment with new options, palettes and colors to recreate previous behaviour. Yes, it is possible to do so.
Solid Glitter may be removed in the future as it can be recreated with Glitter alone.

clayclai, 2023-08-18:
Thank you for the quick reply,

Clay Claiborne, Director
Vietnam: American Holocaust <http://VietnamAmericanHolocaust.com>
Linux Beach Productions
Venice, CA 90291
(310) 581-1536

Read my blogs at the Linux Beach <http://claysbeach.blogspot.com/>
<http://wlcentral.org/user/2965/track>

softhack007, 2023-09-17:
@blazoncek should we close this one as "expected behaviour", or is there something to be fixed for 0.14.0 ?

blazoncek, 2023-09-17:
IMO Solid glitter is on the way to be deprecated as it can be re-created with regular Glitter and Glitter can be *persuaded* to work in the same way it did in 0.13.
Steps:
- select Glitter
- set color slot 1 to be "background" color
- set color slot 3 (Gl) to be "glitter" color
- select palette "* Color 1"

IMO we should close as "not planned".

softhack007, 2023-09-17:
Thanks @blazoncek, I agree. Let's close this one.

closed 2023-09-17 by softhack007

clayclai, 2023-09-17:
In as much as the problem appeared to be the failure of the newer version
to read some color palette entries written in presets by the older version
(b/c they weren't there?), and the solution was to simply recreate those
presets, this matter should be closed.

Thanks for the help.

Clay Claiborne
Linux Beach Productions
Venice, CA 90291
(818)964-0860

Read my blogs at the Linux Beach <http://claysbeach.blogspot.com/>
<http://wlcentral.org/user/2965/track>