wled/WLED #3324: 14.0-b4: Can´t disable shuffle on playlists

gonzabrusco, 2023-08-16:
### What happened?

It seems that in 14.0-b4 the shuffle option in Playlists is always activated. I wanted to make an specific sequence of effects but the playlists keeps being shuffled. I can’t uncheck "Shuffle" option. When you uncheck it and press Save, if you reload the Playlist the Shuffle option will be still checked. 

I'm new to WLED but this seems like a bug. Can anybody confirm?

Thanks

### To Reproduce Bug

Just create a Playlist not shuffled. The result will be a shuffled playlist.

### Expected Behavior

Respect the playlist order. 

### Install Method

Binary from WLED.me

### What version of WLED?

14.0-b4

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2023-08-16:
We have reported the same few days ago but seem the issue is fixed if you use the nightly builds from here https://wled-install.github.io/

blazoncek, 2023-08-16:
I am able to save unrandomised playlist.

softhack007, 2023-08-16:
A workaround is to unckeck "repeat indefinitely", then unckeck "shuffle", and save.

blazoncek, 2023-08-16:
> It seems to be related to "repeat indefinitely"; if shuffle is unchecked but repeat ind. checked, then the next time you open the presets in UI, it shows "shuffle" checked.

That is a lot more information. Thanks.

gonzabrusco, 2023-08-16:
> A workaround is to unckeck "repeat indefinitely", then unckeck "shuffle", and save.

But if I need to "repeat indefinitely", how do I make it not shuffled?

blazoncek, 2023-08-16:
> But if I need to "repeat indefinitely", how do I make it not shuffled?

Edit JSON.

commit bf452e92 (2023-08-16): Playlist saving bugfix.
Fixes #3324