wled/WLED #3327: Build-task issue with VS Code and PlatformIO

arittner, 2023-08-16:
### What happened?

The `PlatformIO: Build` in VS Code is out of order. 

When I click on the `Checkmark/Build` button, the command palette opens and no build is started. When I remove `PlatformIO:` I get this list:

[image]

`Build: binary only` is working. 

First, I thought it's a documentation issue (see https://github.com/Aircoookie/WLED-Docs/issues/150)

VS Code: 1.81.1
OS: Windows 11 x64
PlatformIO: v3.3.1

### To Reproduce Bug

When I click on the `Checkmark/Build` button, the command palette opens and no build is started.

### Expected Behavior

The build should start (at least `Build: binary only`)

### Install Method

Self-Compiled

### What version of WLED?

0.13.3 (build 2208222)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

^- installed method, version and board are irrelevant.

Wiki issue: https://github.com/Aircoookie/WLED-Docs/issues/150

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2023-08-16:
Yeah it was working before but a while back it not longer works , still works if we select run build task from menu ( on mac )  
[image]

blazoncek, 2023-08-17:
When a drop-down list opens after clicking "Build" button (checkmark) just delete pre-filled content of the field. Then select "Build: binary only" from the drop-down list.

dosipod, 2023-08-17:
Delete works 
[image]

softhack007, 2023-08-17:
> 0.13.3 (build 2208222)

If you want to compile old versions of the source code, you need this patch 
https://github.com/Aircoookie/WLED/commit/bc90309dd69533ff9f4b879fb82ecfd1b1db35d7

It fixes compatibility problem with newer versions of platformIO.

blazoncek, 2023-09-28:
Closing as this may not be WLED related issue but MS VSC + PIO issue.

closed 2023-09-28 by blazoncek