wled/WLED #3328: Button states available for HA

Ubercic, 2023-08-16:
Hello,

it would be nice if you find the time to add the option for WLED to share its button states with Home Assistant for use in HA automations.  In my case this would come helpful to control all the bedroom lights with buttons on my WLED night light, instead of having another esp with buttons running ESPhome next to it.

hope you see this idea as usefull as me. if not, feel free to delete the request.

dosipod, 2023-08-16:
It is available via MQTT which you could use in HA  
[image]

Ubercic, 2023-08-16:
Thank you for the tip! Will try that out.

blazoncek, 2023-08-16:
That is available as standard feature. You need to enable it in Sync settings though.
Closing.

closed 2023-08-16 by blazoncek

andymetcalfe93, 2025-03-09:
Hi,

Sorry to dig this one up but it's the only mention I can find of the exact thing I want to do. And I cannot find any other information about it!

I'm new to WLED and HA. I have a ESP32 and have installed WLED on it. I have HA running and have successfully setup the ESP/WLED device.

I have some physical buttons wired up to some of the ESP GPIO pins, and I have set these up as Buttons in the WLED settings. What I want to do to trigger some HA automations using the buttons, however in HA I cannot find a relevant sounding Trigger listed in the Device Trigger settings (the closest one is WLED Restart has been pressed).

I see that above it says that MQTT can be used. @blazoncek I was just wondering that when you say "That is available as standard feature", are you referring to using MQTT as dosipod said, or should this be possible without?

Thanks so much

Andy

blazoncek, 2025-03-09:
Please use WLED [forum](https://wled.discourse.group) or [Discord](https://discord.gg/FMvz8nzmat) for help and support questions.