wled/WLED #3331: Cyceling though presets only works partly

reald, 2023-08-17:
### What happened?

I have several presets defined. When switching to next presets it counts till 5 and then wraps to 1 instead of using number 6.

### To Reproduce Bug

Set preset 3.
Call http://[ip]/win&PL=~
Preset 4 is used
Call http://[ip]/win&PL=~
Preset 5 is used
Call http://[ip]/win&PL=~
Preset 1 is used instead of preset 6

### Expected Behavior

Preset 6 is used because there are more than 5.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b4

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

reald, 2023-08-17:
[presets.zip](https://github.com/Aircoookie/WLED/files/12374060/presets.zip)
Used preset config export

blazoncek, 2023-08-18:
Not a bug.
HTTP API requires full command set. In this case it is: `win&PL=~&P1=1&P2=6`

closed 2023-08-18 by blazoncek