wled/WLED #3334: HA Failed to call service

jordanhinks, 2023-08-20:
### What happened?

When changing the for example the brightness or color state from Home Assistant the following error appears: `Failed to call service light/turn_on. unexpected character: line 1 column 870 (char869)`

However, the service is otherwise called without issue as the requested brightness for example would change without issue.

Within Home Assistant, the Logbook for the entity shows the following: `Office - Shelf Left Firmware became unknown` at the same time an action such as brightness/color is triggered.

### To Reproduce Bug

I have 3 physical LED outputs being used which have the following config, this would apply for GPIO5, GPIO4, and GPIO12:
LED Type: SK6812 RGBW
Length: 27
Auto-calculate white channel from RGB: Accurate

Option for `Make a segment for each output` is ticked.

Power supply: 5v, 3a

I have tested this on 2 separate WEMOs as I have 2 setup to do the same thing but are not connected together in any way.

Below is the error along with showing for example only triggering one segment which causes the error.

[image]

### Expected Behavior

No error to appear.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b4 (build 2308110)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
Home Assistant LOG:
Logger: homeassistant.components.websocket_api.http.connection
Source: helpers/aiohttp_client.py:73
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 19 August 2023 at 21:51:19 (101 occurrences)
Last logged: 12:56:17

[547110223424] unexpected character: line 1 column 870 (char 869)
[547110223424] unexpected character: line 1 column 869 (char 868)
[547110223424] unexpected character: line 1 column 868 (char 867)
[547475498048] unexpected character: line 1 column 869 (char 868)
[547475498048] unexpected character: line 1 column 868 (char 867)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 226, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1974, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2011, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 870, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 942, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 591, in async_handle_light_off_service
    await light.async_turn_off(**filter_turn_off_params(light, params))
  File "/usr/src/homeassistant/homeassistant/components/wled/helpers.py", line 28, in handler
    await func(self, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/components/wled/light.py", line 87, in async_turn_off
    await self.coordinator.wled.master(on=False, transition=transition)
  File "/usr/local/lib/python3.11/site-packages/wled/wled.py", line 368, in master
    await self.request("/json/state", method="POST", data=state)
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/wled/wled.py", line 211, in request
    response_data = await response.json()
                    ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/aiohttp_client.py", line 73, in json
    return await super().json(*args, loads=loads, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/client_reqrep.py", line 1120, in json
    return loads(stripped.decode(encoding))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
orjson.JSONDecodeError: unexpected character: line 1 column 869 (char 868)

____________

Logger: homeassistant
Source: helpers/aiohttp_client.py:73
First occurred: 19 August 2023 at 22:04:33 (24 occurrences)
Last logged: 12:53:38

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1199, in async_request_call
    return await coro
           ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 942, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 580, in async_handle_light_on_service
    await light.async_turn_on(**filter_turn_on_params(light, params))
  File "/usr/src/homeassistant/homeassistant/components/wled/helpers.py", line 28, in handler
    await func(self, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/components/wled/light.py", line 97, in async_turn_on
    await self.coordinator.wled.master(
  File "/usr/local/lib/python3.11/site-packages/wled/wled.py", line 368, in master
    await self.request("/json/state", method="POST", data=state)
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/wled/wled.py", line 211, in request
    response_data = await response.json()
                    ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/aiohttp_client.py", line 73, in json
    return await super().json(*args, loads=loads, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/client_reqrep.py", line 1120, in json
    return loads(stripped.decode(encoding))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
orjson.JSONDecodeError: unexpected character: line 1 column 868 (char 867)
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2023-08-20:
Yeah I faced the same error few days ago also on esp8266  that is was trying to demo for someone and it was reported in discord in HA channel which is kinda dead    , i did not see other impact  
[image]

blazoncek, 2023-08-20:
Please report HA issues to HA team on their Github.
If there is an issue with WLED they will open appropriate issue here.
Closing.

closed 2023-08-20 by blazoncek

jordanhinks, 2023-08-20:
Is this not considered WLED though considering it happens from the latest beta version of WLED?

dosipod, 2023-08-20:
> Is this not considered WLED though considering it happens from the latest beta version of WLED?

If you see my unit it was not on latest beta . The last few updates to HA have broken other things some of which is not even related to wled . I think this among other issues are related to HA updates which i seen on two PIs

jordanhinks, 2023-08-20:
TY, will open ticket in HA for them to know of this.

dosipod, 2023-08-20:
> Please report HA issues to HA team on their Github. If there is an issue with WLED they will open appropriate issue here. Closing.

Might be best to keep open and mark as external so others would not raise the issue again . Closed issues are rarely seen

jordanhinks, 2023-08-20:
I've now opened this with Home Assistant - https://github.com/home-assistant/supervisor/issues/4499

blazoncek, 2023-08-20:
AFAIK beta versions may not be fully supported in HA.
0.14 made a number of changes to API (not all breaking) so you may need to wait for HA update.