wled/WLED #3335: DDP end of connection isn't detected reliable

koshisan, 2023-08-21:
### What happened?

I have 5 WLED instances based on esp32 running in my living room. The main application is to sync the lighting to my TV through Hyperion. While especially switching it on/off via Home Assistant is sometimes clunky, the sync does work pretty much perfect when changing the settings in Hyperion.

However - when I shut down my system Hyperion shuts down, too. In theorey it should turn off the WLED instances - at least that is how its configured. However, it does not always work. Sometimes - quite randomly - some instances just stay active with the last sync state before shutdown.

This could very well be to my wifi and WLED missing an command, etc. - so not necessarily a bug. 

However, the real problem is, that WLED never seems to recognize that the host isn't streaming anymore. Even if I let it sit for the night and come back in the morning, the LEDs are still turned on and the webfrontend says "WLED is receiving live DDP data" - which is obviously a lie since the host has been powered off for ~8h at that time ;)

To make things worse: I cannot really override it. While the webfrontend presents the option, it doesn't reallly work. I can get passed the message and I can adjust various things, but they don't reflect in the actual hardware. The light always keeps the the latest synced state from when Hyperion was still running. The only workaround is a power cycle of WLED or starting Hyperion again. 

### To Reproduce Bug

Start sync with Hyperion, then shut it down - in some cases WLED continues to think it would still receive live data.

### Expected Behavior

I would expect WLED to close the DDP connection after a timeout when not receiving updates for a while.

Even if this works as intended, the overrride should work under any circumstances and be able to actually shut down the lights...

### Install Method

Binary from WLED.me

### What version of WLED?

esp32 v3.3.6-16

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-08-21:
Hyperion (usually) uses raw RGB data protocol with WLED and engages indefinite realtime mode.
To exit such realtime mode explicit *exit* command has to be provided to WLED from the sender.

Even if you switched to DDP output in Hyperion it may still engage indefinite realtime mode. Otherwise WLED resumes its previous operation after 2.5s of inactive DDP stream.

Not a WLED bug.

For further help and support please use WLED [forum](https://wled.discourse.group) or Discord.
Closing.

closed 2023-08-21 by blazoncek

blazoncek, 2023-08-21:
`{"live":false}` will end realtime mode.

[KB](https://kno.wled.ge/interfaces/json-api/)

koshisan, 2023-08-21:
Thank you for the link - but according to the description "If set to true, enters realtime mode and blanks the LEDs. The realtime timeout option does not have an effect when this command is used, WLED will stay in realtime mode **until** the state (color/effect/segments, excluding brightness) is changed. It is expected that {"live":false} is sent once live data sending is terminated. Not included in state response." realtime mode _should_ also end when I set the effect to solid? Should it not also end on an explicit "off" command in that case?