wled/WLED #3336: Improve 'sleep' button behaviour for WiZ remote

freakintoddles2, 2023-08-21:
**The problem**
When I press the Moon (sleep?) button on the WiZ remote, the brightness is reduced very low which is great. However, the only way to exit this mode and return to normal brightness levels is by pressing one of the 1/2/3/4 buttons. This is not intuituve.

**Describe the solution you'd like**
I propose the following changes to improve usability of sleep mode button:
While in sleep mode:
- Pressing the 'sleep' button again will return to the previous brightness before entering sleep mode, or just ~50% or so if that's simpler.
- Pressing the brightness up button will work to increase the brightness and exit sleep mode
- The 1/2/3/4 buttons will continue to work as they do now and will exit the 'sleep' mode

**Describe alternatives you've considered**
n/a

**Additional context**
n/a

Thank you for your ideas for making WLED better!

blazoncek, 2023-08-23:
I will tag @ctsims as he originally developed remote support.

FYI I am currently adding more global ESP-NOW support and it will include changes to Wiz Mote support.

ctsims, 2023-08-24:
@freakintoddles2 - Thanks for submitting this. 

I definitely agree that the behavior is unintuitive. The current implementation was chosen to mimic the behavior of the Wiz brand smart light products. That means that WLED lights which are using the same Wizmote remote as those smart lights will act the same as the lights around them. I'm honestly not sure why they chose to make sleep mode totally unreactive to the brightness buttons.

I agree that it's not ideal behavior for a standalone light. In the future one thing I've considered is making the light be able listen to multiple remotes and to pick and choose button behaviors from each one (so the light may listen to all events from one remote, but only, say, 'on' and 'off' from another). If we ever get to a point where there's a button mapping available, I think it would be great to be able to choose between "WIZ" sleep mode and one that's less confusing.

blazoncek, 2023-08-24:
You can reuse JSON remote principle from IR.
The only issue I see may be the lack of specification of WiZ mote packets which seem to have different content based on the version of remote (#3341 )

freakintoddles2, 2023-09-18:
@blazoncek any chance this update will be updated in v0.14.0.-b5?

blazoncek, 2023-09-19:
> @blazoncek any chance this change will be updated in v0.14.0.-b5?

No, it will not. 0.15 will have it.

blazoncek, 2023-09-28:
JSON (and better night light handling) added to 0_15 branch. Follow #3341 for updates.

closed 2023-09-28 by blazoncek