wled/WLED #3338: Microphone doesn't work on Wemos Lolin32 Lite

edo44, 2023-08-22:
### What happened?

I plugged a GY-MAX4466 that does work well with the esp32 dev module on every adc1 port. With the Lolin32 Lite the microphone doesn't work on any pin and it doesn't make sense. The module and the microphone is working correctly, I can read the microphone analog read fine. So why is that?

### To Reproduce Bug

Just have a lolin32 board

### Expected Behavior

the microphone should work like on the dev module

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP32-S2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-08-22:
For best results pleas use I2S microphone such as INMP441.

edo44, 2023-08-22:
> 

I have an ADMP401, should I try that too?

blazoncek, 2023-08-22:
IDK. not every microphone is supported.
Check [KB](https://kno.wled.ge/advanced/audio-reactive/).

softhack007, 2023-08-22:
> ### What version of WLED?
> 0.13.3
> 
> ### Which microcontroller/board are you seeing the problem on?
> ESP32-S2

Can verify that above information is correct? If it is, there are some "impossibles" unfortunately:
* ESP32-S2 does not support analog microphones. Only I2S (like IMNP441) is possible.
* WLED 0.13.3 does not have audioreactive, neither does that version support ESP32-S2.

Please clarify. Thanks and kind regards.

dosipod, 2023-08-24:
I am not sure if you solved it now but I can assure you that  Lolin32 Lite boards will work with wled and WLED-SR 0.13.3 with both analog and digital mics because i have them running at the moment on multiple instances  and I also know other guys who are doing the same . For future reference  if you are trying to debug a mic then just use it first with a bare board without any leds and power the MCU from usb  and check peek and wled will not fail you . That will take away wiring and power issues from the equation

softhack007, 2023-09-02:
> I think the problem is the 12V power supply that generates a lot of noise.

Closing, as it looks like the problem was understood and nothing we can solve in software.

closed 2023-09-02 by softhack007