wled/WLED #3339: Change presets.json logic to not need to repeat {"stop":0} 1,000's of times

freakintoddles2, 2023-08-22:
**Is your feature request related to a problem? Please describe.**
When saving and updating presets, it starts taking longer and longer, and corrupted saves more common the more presets I have.  I notices in the json that {"stop":0}, is repeated a LOT. In my current presets.json where I have around ~50 saved presets, it is repeated a whopping 1,170 times!

**Describe the solution you'd like**
I appreciate all the work you all put into this, but I have to imagine there is a more elegant way to do whatever {"stop":0}, is doing without repeating it so many times in the json. This would help cut down the size of the presets.json file a lot, and reduce the error rate when saving new presets.

**Describe alternatives you've considered**
n/a

**Additional context**
n/a

Thank you for your ideas for making WLED better!

softhack007, 2023-08-23:
> imagine there is a more elegant way to do whatever {"stop":0}, is doing without repeating it so many times in the json. This would help cut down the size of the presets.json file a lot, and reduce the error rate when saving new presets.

Hi, I can understand that you find this confusing, however - and @blazoncek might be able to explain better - it also makes the code for reading and managing presets (and segments) very efficient, and it avoids uninitialized data.

To put it into relation: if you only have a single segment, there will be 31 instances of stop:0, so 31 x 11 bytes = 341 bytes. Assume you have 64 presets, it will be around 22 kBytes for stop markers. So I don't see a big win, as the presets file is stored on filesystem and there is plenty of space left; also writing out to flash does not happen too often, so there is no performance bottleneck. 

In fact we could reduce size and improve performance by simply using compressed files. However you would need to un-compress them in order to view plain text.

Maybe I'm missing somesometing?

blazoncek, 2023-08-23:
Each `{"stop":0}` is needed to deactivate/remove a segment. Since 32 is a maximum number of segments ESP32 supports (and 16 for ESP8266) there may be up to 31 (or 15) "stop" commands in each preset.
If you do not want "stop" to appear in your preset just check (select) "Selected segments only" when saving current state as a preset. This may have an undesired effect on your segment set-up if you change them in your presets.

And if your imagination is good then please add the code (and PR) that will achieve the same functionality without needing so many "stop"s.

BTW you can reduce the number of "stop"s by reducing the maximum number of segments by custom compiling with `MAX_NUM_SEGMENTS` set to a desired value.

blazoncek, 2023-08-23:
If you recall 😉 internally there are no "segment IDs". 😁

closed 2024-06-23 by blazoncek