wled/WLED #3341: Issues with WIZ RC Remote Control

moonilightshaker, 2023-08-23:
### What happened?

It seems that there are different versions of the WIZ RC Remote. Enclosed is a quick summary 

### To Reproduce Bug

The Wizmote remote control from REV does not work with Wled. The MAC address is recognized by WLED. But controlling does not work. Tested with ESP32 and ESP8266. Batteries are new and high quality ones.

[image]

### Expected Behavior

Link to my working Remote: https://www.amazon.de/dp/B08HHYKP5F

Photo:
[image]
[image]

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b4

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Link to the not working Remote: https://www.amazon.de/dp/B08DRMHZ76

The packaging is also different from the working remote control

Photos from the not working Remote: 

[image]

[image]

[image]

[image]

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-08-23:
Hi, I would not strictly call it a bug, maybe the WIZ RC is just sending in a different format. 

To find out, you need a debug build (-DWLED_DEBUG), and then watch serial monitor for messages like "unknown incoming ESP Now message".

blazoncek, 2023-08-23:
I've searched for quite a while today and there are 0 bits of information on the WiZ mote message content.
So if they have 10 different versions of message packet, 9 will not work ATM.

blazoncek, 2023-08-23:
I am in a process of updating and extending ESP-NOW support but I lack WiZ mote remote to test.

moonilightshaker, 2023-08-23:
Thank you fot the response. 

When i Debug it, this message comes up:
Unknown incoming ESP Now message received of length 11

blazoncek, 2023-08-23:
Current code expects the length to be 13.

moonilightshaker, 2023-08-23:
Oh Ok. 
Is there anything else I can do to help?

blazoncek, 2023-08-23:
Send a remote? 😁 
Or dump a message.

moonilightshaker, 2023-08-23:
> Send a remote? 😁 Or dump a message.

If you give me your address I will send you one.

moonilightshaker, 2023-08-24:
> Please contact me on Discord.

Done (moon....)

ctsims, 2023-08-24:
Thanks for reporting! Definitely annoying for them to change up the formatting. The original message contents were definitely pieced together (rather than published) so we'll need to reverse engineer the structure again to add support for the other message format.

Unfortunately the REV remotes don't look available in the US, but it's probably fairly straightforward to piece together the packet structure if you could provide some debug outputs. Would it be easier to get a standalone ESP build to get console data out of, or easier for you to get a build of WLED which dumps the data?

moonilightshaker, 2023-08-24:
> Thanks for reporting! Definitely annoying for them to change up the formatting. The original message contents were definitely pieced together (rather than published) so we'll need to reverse [...]

Do you have a project that I can flash to an ESP32? A standalone ESP build would be fine. Then I could send you the output. 
I have not dealt with WLAN sniffer yet.

ctsims, 2023-08-24:
Great! Yeah, I should be able to send over a build for ESP32 that you can flash that'll dump the right info to the console in an easily consumable way. No need to try to sniff the packets out of the air, the WIFI frames that ESP-NOW uses are so uncommon that you'd have a hell of a time getting something compatible to see them. 

Will try and get you something tonight to test with.

ctsims, 2023-08-25:
@moonilightshaker Ok, here's a quick [binary](https://drive.google.com/file/d/1BPzUe4j7xmeuPeXIpXChCGFMsH_cDXYB/view?usp=drive_link) that you can flash onto an esp32 that should set up the access point and listen for incoming packets and dump them to hex. 

If you can grab a few packets and just mention which ones are associated with what button presses, that would help immensely.  Please make sure to include the full trace, part of the payload is an incrementing nonce, so having the incoming messages listed in order is super helpful (there may be duplicates, still helpful to include them). 

In case there are any issues with the binary this is the [simple sketch](https://gist.github.com/ctsims/3522d88029c8d49710cc6686702b9494) since sometimes different platforms have different compilation issues. 

Thanks again for being willing to help out with it! Hopefully we can get this resolved easily.

moonilightshaker, 2023-08-25:
@ctsims 
Enclosed are the outputs of the ON key. Is this OK? The HEX code changes with each keystroke.

_rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1324
ho 0 tail 12 room 4
load:0x40078000,len:13508
load:0x40080400,len:3604
entry 0x400805f0
Inititalizing esp now_

_Bytes received: 11 
80CD0008ACA54628DA3
Bytes received: 11 
80CD0008ACA54628DA3
Bytes received: 11 
80CD0008ACA54628DA3
Bytes received: 11 
80CD0008ACA54628DA3
Bytes received: 11 
80CD0008ACA54628DA3
Bytes received: 11 
80CD0008ACA54628DA3
Bytes received: 11 
80CD0008ACA54628DA3
Bytes received: 11 
80CD0008ACA54628DA3
Bytes received: 11 
80CD0008ACA54628DA3
Bytes received: 11 
80CD0008ACA54628DA3
Bytes received: 11 
80CD0008ACA54628DA3_

ctsims, 2023-08-25:
@moonilightshaker Thanks for sending!

To clarify: is the output you sent with 11 packets the result of a single button press or multiple times pressing the button? (on my wizmotes, it blasts a number of packets for every button press). In general the packets on the other remotes change each time you press the same button as a nonce increments, so it's great to get the result of those changes so I can tell where the nonce is in this packet structure.

It'd be helpful to have a few more data points. Can you post traces with the following?
- Output from pressing the same button (say, the "dim" button) multiple times in a row
- Output from a few other buttons (can be any of them)

I forgot to mention but on the common wizmote we both have the 'on' button is a little special, so having traces from the non-on buttons is really helpful to differentiate.

moonilightshaker, 2023-08-25:
@ctsims The Output is from a Single Button press.

Output from the "Dim - " button. Pressed 20 times in a row:
_Bytes received: 11 
80D4000B6BEB46AD13D
Bytes received: 11 
80D4000B6BEB46AD13D
Bytes received: 11 
80D4000B6BEB46AD13D
Bytes received: 11 
80D4000B6BEB46AD13D
Bytes received: 11 
80D4000B6BEB46AD13D
Bytes received: 11 
80D4000B6BEB46AD13D
Bytes received: 11
80D5000598139EB5D8
Bytes received: 11
80D5000598139EB5D8
Bytes received: 11
80D5000598139EB5D8
Bytes received: 11
80D5000598139EB5D8
Bytes received: 11
80D5000598139EB5D8
Bytes received: 11
80D5000598139EB5D8
Bytes received: 11
80D5000598139EB5D8
Bytes received: 11
80D5000598139EB5D8
Bytes received: 11
80D5000598139EB5D8
Bytes received: 11 
80D60006074FBB862FE
Bytes received: 11 
80D60006074FBB862FE
Bytes received: 11 
80D60006074FBB862FE
Bytes received: 11
80D60006074FBB862FE
Bytes received: 11
80D60006074FBB862FE
Bytes received: 11
80D60006074FBB862FE
Bytes received: 11
80D60006074FBB862FE
Bytes received: 11
80D7000A5895AB27C30
Bytes received: 11
80D7000A5895AB27C30
Bytes received: 11
80D7000A5895AB27C30
Bytes received: 11
80D7000A5895AB27C30
Bytes received: 11
80D7000A5895AB27C30
Bytes received: 11
80D7000A5895AB27C30
Bytes received: 11
80D7000A5895AB27C30
Bytes received: 11
80D7000A5895AB27C30
Bytes received: 11
80D80006F762C5608B
Bytes received: 11
80D80006F762C5608B
Bytes received: 11
80D80006F762C5608B
Bytes received: 11
[... 214 lines cut ...]
80E60001D8CFB7D02D
Bytes received: 11
80E60001D8CFB7D02D
Bytes received: 11
80E60001D8CFB7D02D
Bytes received: 11
80E60001D8CFB7D02D
Bytes received: 11
80E60001D8CFB7D02D
Bytes received: 11
80E60001D8CFB7D02D
Bytes received: 11
80E700016B6C859CAC
Bytes received: 11
80E700016B6C859CAC
Bytes received: 11
80E700016B6C859CAC
Bytes received: 11
80E700016B6C859CAC
Bytes received: 11
80E700016B6C859CAC
Bytes received: 11
80E700016B6C859CAC
Bytes received: 11
80E700016B6C859CAC_

moonilightshaker, 2023-08-25:
Output from the "1" button. Pressed 10 times in a row:
_Bytes received: 11
80E8000DBBC615C56B
Bytes received: 11
80E8000DBBC615C56B
Bytes received: 11
80E8000DBBC615C56B
Bytes received: 11
80E8000DBBC615C56B
Bytes received: 11
80E8000DBBC615C56B
Bytes received: 11
80E8000DBBC615C56B
Bytes received: 11
80E8000DBBC615C56B
Bytes received: 11
80E8000DBBC615C56B
Bytes received: 11
80E8000DBBC615C56B
Bytes received: 11
80E8000DBBC615C56B
Bytes received: 11
80E8000DBBC615C56B
Bytes received: 11
80E9000AF711C6EF5A
Bytes received: 11
80E9000AF711C6EF5A
Bytes received: 11
80E9000AF711C6EF5A
Bytes received: 11
80E9000AF711C6EF5A
Bytes received: 11
80E9000AF711C6EF5A
Bytes received: 11
80E9000AF711C6EF5A
Bytes received: 11
80E9000AF711C6EF5A
Bytes received: 11
80E9000AF711C6EF5A
Bytes received: 11
80E9000AF711C6EF5A
Bytes received: 11
80E9000AF711C6EF5A
Bytes received: 11
80EA00055EBD3A28FD4
Bytes received: 11
80EA00055EBD3A28FD4
Bytes received: 11
80EA00055EBD3A28FD4
Bytes received: 11
80EA00055EBD3A28FD4
Bytes received: 11
80EA00055EBD3A28FD4
Bytes received: 11
80EA00055EBD3A28FD4
Bytes received: 11
80EA00055EBD3A28FD4
Bytes received: 11
80EA00055EBD3A28FD4
Bytes received: 11
80EA00055EBD3A28FD4
Bytes received: 11
80EA00055EBD3A28FD4
Bytes received: 11
80EA00055EBD3A28FD4
Bytes received: 11
80EB000ACC05A852A26
Bytes received: 11
80EB000ACC05A852A26
Bytes received: 11
80EB000ACC05A852A26
Bytes received: 11
80EB000ACC05A852A26
Bytes received: 11
80EB000ACC05A852A26
Bytes received: 11
80EB000ACC05A852A26
Bytes received: 11
80EB000ACC05A852A26
Bytes received: 11
80EB000ACC05A852A26
Bytes received: 11
80EB000ACC05A852A26
Bytes received: 11
80EB000ACC05A852A26
Bytes received: 11
80EB000ACC05A852A26
Bytes received: 11
80EB000ACC05A852A26
Bytes received: 11
80EC000322C2B668F9D
Bytes received: 11
80EC000322C2B668F9D
Bytes received: 11
80EC000322C2B668F9D
Bytes received: 11
80EC000322C2B668F9D
Bytes received: 11
80EC000322C2B668F9D
Bytes received: 11
80EC000322C2B668F9D
Bytes received: 11
80EC000322C2B668F9D
Bytes received: 11
80EC000322C2B668F9D
Bytes received: 11
80EC000322C2B668F9D
Bytes received: 11
80EC000322C2B668F9D
Bytes received: 11
80EC000322C2B668F9D
Bytes received: 11
80ED0007330A3D2B82
Bytes received: 11
80ED0007330A3D2B82
Bytes received: 11
80ED0007330A3D2B82
Bytes received: 11
80ED0007330A3D2B82
Bytes received: 11
80ED0007330A3D2B82
Bytes received: 11
80ED0007330A3D2B82
Bytes received: 11
80ED0007330A3D2B82
Bytes received: 11
80ED0007330A3D2B82
Bytes received: 11
80ED0007330A3D2B82
Bytes received: 11
80ED0007330A3D2B82
Bytes received: 11
80ED0007330A3D2B82
Bytes received: 11
80EE000ADAF9F9AEFEB
Bytes received: 11
80EE000ADAF9F9AEFEB
Bytes received: 11
80EE000ADAF9F9AEFEB
Bytes received: 11
80EE000ADAF9F9AEFEB
Bytes received: 11
80EE000ADAF9F9AEFEB
Bytes received: 11
80EE000ADAF9F9AEFEB
Bytes received: 11
80EE000ADAF9F9AEFEB
Bytes received: 11
80EE000ADAF9F9AEFEB
Bytes received: 11
80EE000ADAF9F9AEFEB
Bytes received: 11
80EE000ADAF9F9AEFEB
Bytes received: 11
80EE000ADAF9F9AEFEB
Bytes received: 11
80EF00048CC9442812
Bytes received: 11
80EF00048CC9442812
Bytes received: 11
80EF00048CC9442812
Bytes received: 11
80EF00048CC9442812
Bytes received: 11
80EF00048CC9442812
Bytes received: 11
80EF00048CC9442812
Bytes received: 11
80EF00048CC9442812
Bytes received: 11
80EF00048CC9442812
Bytes received: 11
80EF00048CC9442812
Bytes received: 11
80EF00048CC9442812
Bytes received: 11
80EF00048CC9442812
Bytes received: 11
80EF00048CC9442812
Bytes received: 11
80F0000D9B33C757554
Bytes received: 11
80F0000D9B33C757554
Bytes received: 11
80F0000D9B33C757554
Bytes received: 11
80F0000D9B33C757554
Bytes received: 11
80F0000D9B33C757554
Bytes received: 11
80F0000D9B33C757554
Bytes received: 11
80F0000D9B33C757554
Bytes received: 11
80F0000D9B33C757554
Bytes received: 11
80F0000D9B33C757554
Bytes received: 11
80F0000D9B33C757554
Bytes received: 11
80F0000D9B33C757554
Bytes received: 11
80F0000D9B33C757554
Bytes received: 11
80F0000D9B33C757554
Bytes received: 11
80F1000D37B1B55FDF8
Bytes received: 11
80F1000D37B1B55FDF8
Bytes received: 11
80F1000D37B1B55FDF8
Bytes received: 11
80F1000D37B1B55FDF8
Bytes received: 11
80F1000D37B1B55FDF8
Bytes received: 11
80F1000D37B1B55FDF8
Bytes received: 11
80F1000D37B1B55FDF8
Bytes received: 11
80F1000D37B1B55FDF8
Bytes received: 11
80F1000D37B1B55FDF8
Bytes received: 11
80F1000D37B1B55FDF8
Bytes received: 11
80F1000D37B1B55FDF8
Bytes received: 11
80F1000D37B1B55FDF8
Bytes received: 11
80F1000D37B1B55FDF8
Bytes received: 11
80F1000D37B1B55FDF8_

moonilightshaker, 2023-08-25:
Output from the "off" button. Pressed 10 times in a row:
_Bytes received: 11
80FC00091F089DB3FE
Bytes received: 11
80FC00091F089DB3FE
Bytes received: 11
80FC00091F089DB3FE
Bytes received: 11
80FC00091F089DB3FE
Bytes received: 11
80FC00091F089DB3FE
Bytes received: 11
80FC00091F089DB3FE
Bytes received: 11
80FC00091F089DB3FE
Bytes received: 11
80FC00091F089DB3FE
Bytes received: 11
80FC00091F089DB3FE
Bytes received: 11
80FC00091F089DB3FE
Bytes received: 11
80FC00091F089DB3FE
Bytes received: 11 
80FD0009D4B733DF44B
Bytes received: 11 
80FD0009D4B733DF44B
Bytes received: 11 
80FD0009D4B733DF44B
Bytes received: 11 
80FD0009D4B733DF44B
Bytes received: 11 
80FD0009D4B733DF44B
Bytes received: 11 
80FD0009D4B733DF44B
Bytes received: 11 
80FD0009D4B733DF44B
Bytes received: 11
80FD0009D4B733DF44B
Bytes received: 11
80FD0009D4B733DF44B
Bytes received: 11
80FD0009D4B733DF44B
Bytes received: 11
80FD0009D4B733DF44B
Bytes received: 11
80FE00099F5592E48BB
Bytes received: 11
80FE00099F5592E48BB
Bytes received: 11
80FE00099F5592E48BB
Bytes received: 11
80FE00099F5592E48BB
Bytes received: 11
80FE00099F5592E48BB
Bytes received: 11
80FE00099F5592E48BB
Bytes received: 11
80FE00099F5592E48BB
Bytes received: 11
80FE00099F5592E48BB
Bytes received: 11
80FE00099F5592E48BB
Bytes received: 11
80FE00099F5592E48BB
Bytes received: 11
80FE00099F5592E48BB
Bytes received: 11
80FF0001A98D1B69A2
Bytes received: 11
80FF0001A98D1B69A2
Bytes received: 11
80FF0001A98D1B69A2
Bytes received: 11
80FF0001A98D1B69A2
Bytes received: 11
80FF0001A98D1B69A2
Bytes received: 11
80FF0001A98D1B69A2
Bytes received: 11
80FF0001A98D1B69A2
Bytes received: 11
80FF0001A98D1B69A2
Bytes received: 11
80FF0001A98D1B69A2
Bytes received: 11
80FF0001A98D1B69A2
Bytes received: 11
80FF0001A98D1B69A2
Bytes received: 11
80FF0001A98D1B69A2
Bytes received: 11
800100BACB74264BEA
Bytes received: 11
800100BACB74264BEA
Bytes received: 11
800100BACB74264BEA
Bytes received: 11
800100BACB74264BEA
Bytes received: 11
800100BACB74264BEA
Bytes received: 11
800100BACB74264BEA
Bytes received: 11
800100BACB74264BEA
Bytes received: 11
800100BACB74264BEA
Bytes received: 11
800100BACB74264BEA
Bytes received: 11
8011003E876AB17E27
Bytes received: 11
8011003E876AB17E27
Bytes received: 11
8011003E876AB17E27
Bytes received: 11
8011003E876AB17E27
Bytes received: 11
8011003E876AB17E27
Bytes received: 11
8011003E876AB17E27
Bytes received: 11
8011003E876AB17E27
Bytes received: 11
8011003E876AB17E27
Bytes received: 11
8011003E876AB17E27
Bytes received: 11
8011003E876AB17E27
Bytes received: 11
8011003E876AB17E27
Bytes received: 11
8011003E876AB17E27
Bytes received: 11
8011003E876AB17E27
Bytes received: 11
80210099B1A664DB5
Bytes received: 11
80210099B1A664DB5
Bytes received: 11
80210099B1A664DB5
Bytes received: 11
80210099B1A664DB5
Bytes received: 11
80210099B1A664DB5
Bytes received: 11
80210099B1A664DB5
Bytes received: 11
80210099B1A664DB5
Bytes received: 11
80210099B1A664DB5
Bytes received: 11
80210099B1A664DB5
Bytes received: 11
80210099B1A664DB5
Bytes received: 11
80210099B1A664DB5
Bytes received: 11
80210099B1A664DB5
Bytes received: 11
803100D1A4BE6CA39B
Bytes received: 11
803100D1A4BE6CA39B
Bytes received: 11
803100D1A4BE6CA39B
Bytes received: 11
803100D1A4BE6CA39B
Bytes received: 11
803100D1A4BE6CA39B
Bytes received: 11
803100D1A4BE6CA39B
Bytes received: 11
803100D1A4BE6CA39B
Bytes received: 11
803100D1A4BE6CA39B
Bytes received: 11
803100D1A4BE6CA39B
Bytes received: 11
803100D1A4BE6CA39B
Bytes received: 11
803100D1A4BE6CA39B
Bytes received: 11
803100D1A4BE6CA39B
Bytes received: 11
803100D1A4BE6CA39B
Bytes received: 11
803100D1A4BE6CA39B
Bytes received: 11
80410053CC59AF8B61
Bytes received: 11
80410053CC59AF8B61
Bytes received: 11
80410053CC59AF8B61
Bytes received: 11
80410053CC59AF8B61
Bytes received: 11
80410053CC59AF8B61
Bytes received: 11
80410053CC59AF8B61
Bytes received: 11
80410053CC59AF8B61
Bytes received: 11
80410053CC59AF8B61
Bytes received: 11
80410053CC59AF8B61
Bytes received: 11
80410053CC59AF8B61
Bytes received: 11
80410053CC59AF8B61
Bytes received: 11
80410053CC59AF8B61
Bytes received: 11
80410053CC59AF8B61
Bytes received: 11
805100C9D1CB88D2B1
Bytes received: 11
805100C9D1CB88D2B1
Bytes received: 11
805100C9D1CB88D2B1
Bytes received: 11
805100C9D1CB88D2B1
Bytes received: 11
805100C9D1CB88D2B1
Bytes received: 11
805100C9D1CB88D2B1
Bytes received: 11
805100C9D1CB88D2B1
Bytes received: 11
805100C9D1CB88D2B1
Bytes received: 11
805100C9D1CB88D2B1
Bytes received: 11
805100C9D1CB88D2B1
Bytes received: 11
805100C9D1CB88D2B1
Bytes received: 11
805100C9D1CB88D2B1
Bytes received: 11
805100C9D1CB88D2B1_

moonilightshaker, 2023-08-25:
Output from the "4" button. Pressed 10 times in a row:
_Bytes received: 11
8061000DC2C195A11 
Bytes received: 11
8061000DC2C195A11 
Bytes received: 11
8061000DC2C195A11 
Bytes received: 11
8061000DC2C195A11 
Bytes received: 11
8061000DC2C195A11 
Bytes received: 11
8061000DC2C195A11 
Bytes received: 11
8061000DC2C195A11 
Bytes received: 11
8061000DC2C195A11 
Bytes received: 11
8061000DC2C195A11 
Bytes received: 11
8061000DC2C195A11 
Bytes received: 11
8061000DC2C195A11 
Bytes received: 11
8061000DC2C195A11 
Bytes received: 11
80710029D7F3B392B 
Bytes received: 11
80710029D7F3B392B 
Bytes received: 11
80710029D7F3B392B 
Bytes received: 11
80710029D7F3B392B 
Bytes received: 11
80710029D7F3B392B 
Bytes received: 11
80710029D7F3B392B 
Bytes received: 11
80710029D7F3B392B
Bytes received: 11
80710029D7F3B392B
Bytes received: 11
80710029D7F3B392B
Bytes received: 11
80710029D7F3B392B
Bytes received: 11
80710029D7F3B392B
Bytes received: 11
80710029D7F3B392B
Bytes received: 11
80710029D7F3B392B
Bytes received: 11
80710029D7F3B392B
Bytes received: 11
808100879EAB5EE59A
Bytes received: 11
808100879EAB5EE59A
Bytes received: 11
808100879EAB5EE59A
Bytes received: 11
808100879EAB5EE59A
Bytes received: 11
808100879EAB5EE59A
Bytes received: 11
808100879EAB5EE59A
Bytes received: 11
808100879EAB5EE59A
Bytes received: 11
808100879EAB5EE59A
Bytes received: 11
808100879EAB5EE59A
Bytes received: 11
808100879EAB5EE59A
Bytes received: 11
80910037B17A98B5F
Bytes received: 11
80910037B17A98B5F
Bytes received: 11
80910037B17A98B5F
Bytes received: 11
80910037B17A98B5F
Bytes received: 11
80910037B17A98B5F
Bytes received: 11
80910037B17A98B5F
Bytes received: 11
80910037B17A98B5F
Bytes received: 11
80910037B17A98B5F
Bytes received: 11
80910037B17A98B5F
Bytes received: 11
80910037B17A98B5F
Bytes received: 11
80910037B17A98B5F
Bytes received: 11
80910037B17A98B5F
Bytes received: 11
80910037B17A98B5F
Bytes received: 11
80A1008A96BF1F46EE
Bytes received: 11
80A1008A96BF1F46EE
Bytes received: 11
80A1008A96BF1F46EE
Bytes received: 11
80A1008A96BF1F46EE
Bytes received: 11
80A1008A96BF1F46EE
Bytes received: 11
80A1008A96BF1F46EE
Bytes received: 11
80A1008A96BF1F46EE
Bytes received: 11
80A1008A96BF1F46EE
Bytes received: 11
80A1008A96BF1F46EE
Bytes received: 11
80A1008A96BF1F46EE
Bytes received: 11
80A1008A96BF1F46EE
Bytes received: 11
80B100FCCEDC57906E
Bytes received: 11
80B100FCCEDC57906E
Bytes received: 11
80B100FCCEDC57906E
Bytes received: 11
80B100FCCEDC57906E
Bytes received: 11
80B100FCCEDC57906E
Bytes received: 11
80B100FCCEDC57906E
Bytes received: 11
80B100FCCEDC57906E
Bytes received: 11
80B100FCCEDC57906E
Bytes received: 11
80B100FCCEDC57906E
Bytes received: 11
80B100FCCEDC57906E
Bytes received: 11
80B100FCCEDC57906E
Bytes received: 11
80B100FCCEDC57906E
Bytes received: 11
80B100FCCEDC57906E
Bytes received: 11
80C10059DCF35CEBAA
Bytes received: 11
80C10059DCF35CEBAA
Bytes received: 11
80C10059DCF35CEBAA
Bytes received: 11
80C10059DCF35CEBAA
Bytes received: 11
80C10059DCF35CEBAA
Bytes received: 11
80C10059DCF35CEBAA
Bytes received: 11
80C10059DCF35CEBAA
Bytes received: 11
80C10059DCF35CEBAA
Bytes received: 11
80C10059DCF35CEBAA
Bytes received: 11
80C10059DCF35CEBAA
Bytes received: 11
80D1003C7E3444F079
Bytes received: 11
80D1003C7E3444F079
Bytes received: 11
80D1003C7E3444F079
Bytes received: 11
80D1003C7E3444F079
Bytes received: 11
80D1003C7E3444F079
Bytes received: 11
80D1003C7E3444F079
Bytes received: 11
80D1003C7E3444F079
Bytes received: 11
80D1003C7E3444F079
Bytes received: 11
80D1003C7E3444F079
Bytes received: 11
80D1003C7E3444F079
Bytes received: 11
80D1003C7E3444F079
Bytes received: 11
80D1003C7E3444F079
Bytes received: 11
80D1003C7E3444F079
Bytes received: 11
80E100E1C3CE55B7
Bytes received: 11
80E100E1C3CE55B7
Bytes received: 11
80E100E1C3CE55B7
Bytes received: 11
80E100E1C3CE55B7
Bytes received: 11
80E100E1C3CE55B7
Bytes received: 11
80E100E1C3CE55B7
Bytes received: 11
80E100E1C3CE55B7
Bytes received: 11
80E100E1C3CE55B7
Bytes received: 11
80E100E1C3CE55B7
Bytes received: 11
80E100E1C3CE55B7
Bytes received: 11
80E100E1C3CE55B7
Bytes received: 11
80E100E1C3CE55B7
Bytes received: 11
80F100A2A482DFD116
Bytes received: 11
80F100A2A482DFD116
Bytes received: 11
80F100A2A482DFD116
Bytes received: 11
80F100A2A482DFD116
Bytes received: 11
80F100A2A482DFD116
Bytes received: 11
80F100A2A482DFD116
Bytes received: 11
80F100A2A482DFD116
Bytes received: 11
80F100A2A482DFD116
Bytes received: 11
80F100A2A482DFD116
Bytes received: 11
80F100A2A482DFD116
Bytes received: 11
80F100A2A482DFD116_

moonilightshaker, 2023-08-25:
Output from the "Dim +" button. Pressed 10 times in a row:
_Bytes received: 11
8012100BA7AF7A743C6
Bytes received: 11
8012100BA7AF7A743C6
Bytes received: 11
8012100BA7AF7A743C6
Bytes received: 11
8012100BA7AF7A743C6
Bytes received: 11
8012100BA7AF7A743C6
Bytes received: 11
8012100BA7AF7A743C6
Bytes received: 11
8012100BA7AF7A743C6
Bytes received: 11
8013100CBAC42C972B1
Bytes received: 11
8013100CBAC42C972B1
Bytes received: 11
8013100CBAC42C972B1
Bytes received: 11
8013100CBAC42C972B1
Bytes received: 11
8013100CBAC42C972B1
Bytes received: 11
8013100CBAC42C972B1
Bytes received: 11
8014100DC1DBBED1DFF
Bytes received: 11
8014100DC1DBBED1DFF
Bytes received: 11
8014100DC1DBBED1DFF
Bytes received: 11
8014100DC1DBBED1DFF
Bytes received: 11
8014100DC1DBBED1DFF
Bytes received: 11
8014100DC1DBBED1DFF
Bytes received: 11
8015100466B82A108F
Bytes received: 11
8015100466B82A108F
Bytes received: 11
8015100466B82A108F
Bytes received: 11
8015100466B82A108F
Bytes received: 11
8015100466B82A108F
Bytes received: 11
8015100466B82A108F
Bytes received: 11
8015100466B82A108F
Bytes received: 11
8016100DDE687CA5A17
Bytes received: 11
8016100DDE687CA5A17
Bytes received: 11
8016100DDE687CA5A17
Bytes received: 11
8016100DDE687CA5A17
Bytes received: 11
8016100DDE687CA5A17
Bytes received: 11
8016100DDE687CA5A17
Bytes received: 11
8016100DDE687CA5A17
Bytes received: 11
8016100DDE687CA5A17
Bytes received: 11
80171005D94B931973F
Bytes received: 11
80171005D94B931973F
Bytes received: 11
80171005D94B931973F
Bytes received: 11
80171005D94B931973F
Bytes received: 11
80171005D94B931973F
Bytes received: 11
80171005D94B931973F
Bytes received: 11
8018100FD8E793F36C
Bytes received: 11
8018100FD8E793F36C
Bytes received: 11
8018100FD8E793F36C
Bytes received: 11
8018100FD8E793F36C
Bytes received: 11
8018100FD8E793F36C
Bytes received: 11
8018100FD8E793F36C
Bytes received: 11
8019100CDCF5A34B684
Bytes received: 11
8019100CDCF5A34B684
Bytes received: 11
8019100CDCF5A34B684
Bytes received: 11
8019100CDCF5A34B684
Bytes received: 11
8019100CDCF5A34B684
Bytes received: 11
8019100CDCF5A34B684
Bytes received: 11
8019100CDCF5A34B684
Bytes received: 11
8019100CDCF5A34B684
Bytes received: 11
801A100317D2C7A74EC
Bytes received: 11
801A100317D2C7A74EC
Bytes received: 11
801A100317D2C7A74EC
Bytes received: 11
801A100317D2C7A74EC
Bytes received: 11
801A100317D2C7A74EC
Bytes received: 11
801A100317D2C7A74EC
Bytes received: 11
801A100317D2C7A74EC
Bytes received: 11
801B100CC8994ECDF3
Bytes received: 11
801B100CC8994ECDF3
Bytes received: 11
801B100CC8994ECDF3
Bytes received: 11
801B100CC8994ECDF3
Bytes received: 11
801B100CC8994ECDF3
Bytes received: 11
801B100CC8994ECDF3_

moonilightshaker, 2023-08-25:
Output from the "2" button. Pressed 10 times in a row:
_Bytes received: 11
801C100AC6F8E5C45A
Bytes received: 11
801C100AC6F8E5C45A
Bytes received: 11
801C100AC6F8E5C45A
Bytes received: 11
801C100AC6F8E5C45A
Bytes received: 11
801C100AC6F8E5C45A
Bytes received: 11
801C100AC6F8E5C45A
Bytes received: 11
801C100AC6F8E5C45A
Bytes received: 11
801C100AC6F8E5C45A
Bytes received: 11
801C100AC6F8E5C45A
Bytes received: 11
801C100AC6F8E5C45A
Bytes received: 11
801D100CBD7F875D13E
Bytes received: 11
801D100CBD7F875D13E
Bytes received: 11
801D100CBD7F875D13E
Bytes received: 11
801D100CBD7F875D13E
Bytes received: 11
801D100CBD7F875D13E
Bytes received: 11
801D100CBD7F875D13E
Bytes received: 11
801D100CBD7F875D13E
Bytes received: 11
801D100CBD7F875D13E
Bytes received: 11
801D100CBD7F875D13E
Bytes received: 11
801D100CBD7F875D13E
Bytes received: 11
801D100CBD7F875D13E
Bytes received: 11
801E100236D92DC4429
Bytes received: 11
801E100236D92DC4429
Bytes received: 11
801E100236D92DC4429
Bytes received: 11
801E100236D92DC4429
Bytes received: 11
801E100236D92DC4429
Bytes received: 11
801E100236D92DC4429
Bytes received: 11
801E100236D92DC4429
Bytes received: 11
801E100236D92DC4429
Bytes received: 11
801E100236D92DC4429
Bytes received: 11
801E100236D92DC4429
Bytes received: 11
801F100D762C511B9C
Bytes received: 11
801F100D762C511B9C
Bytes received: 11
801F100D762C511B9C
Bytes received: 11
801F100D762C511B9C
Bytes received: 11
801F100D762C511B9C
Bytes received: 11
801F100D762C511B9C
Bytes received: 11
801F100D762C511B9C
Bytes received: 11
801F100D762C511B9C
Bytes received: 11
801F100D762C511B9C
Bytes received: 11
801F100D762C511B9C
Bytes received: 11
801F100D762C511B9C
Bytes received: 11
80201003941DAB6626
Bytes received: 11
80201003941DAB6626
Bytes received: 11
80201003941DAB6626
Bytes received: 11
80201003941DAB6626
Bytes received: 11
80201003941DAB6626
Bytes received: 11
80201003941DAB6626
Bytes received: 11
80201003941DAB6626
Bytes received: 11
80201003941DAB6626
Bytes received: 11
80201003941DAB6626
Bytes received: 11
8021100F3C65B9B56
Bytes received: 11
8021100F3C65B9B56
Bytes received: 11
8021100F3C65B9B56
Bytes received: 11
8021100F3C65B9B56
Bytes received: 11
8021100F3C65B9B56
Bytes received: 11
8021100F3C65B9B56
Bytes received: 11
8021100F3C65B9B56
Bytes received: 11
8021100F3C65B9B56
Bytes received: 11
8021100F3C65B9B56
Bytes received: 11
8021100F3C65B9B56
Bytes received: 11
8021100F3C65B9B56
Bytes received: 11
80221005468FEDB29C
Bytes received: 11
80221005468FEDB29C
Bytes received: 11
80221005468FEDB29C
Bytes received: 11
80221005468FEDB29C
Bytes received: 11
80221005468FEDB29C
Bytes received: 11
80221005468FEDB29C
Bytes received: 11
80221005468FEDB29C
Bytes received: 11
80221005468FEDB29C
Bytes received: 11
80221005468FEDB29C
Bytes received: 11
80221005468FEDB29C
Bytes received: 11
8023100FFC21078BAA
Bytes received: 11
8023100FFC21078BAA
Bytes received: 11
8023100FFC21078BAA
Bytes received: 11
8023100FFC21078BAA
Bytes received: 11
8023100FFC21078BAA
Bytes received: 11
8023100FFC21078BAA
Bytes received: 11
8023100FFC21078BAA
Bytes received: 11
8023100FFC21078BAA
Bytes received: 11
8023100FFC21078BAA
Bytes received: 11
8023100FFC21078BAA
Bytes received: 11
8023100FFC21078BAA
Bytes received: 11
80241006C6C9EE7DD5
Bytes received: 11
80241006C6C9EE7DD5
Bytes received: 11
80241006C6C9EE7DD5
Bytes received: 11
80241006C6C9EE7DD5
Bytes received: 11
80241006C6C9EE7DD5
Bytes received: 11
80241006C6C9EE7DD5
Bytes received: 11
80241006C6C9EE7DD5
Bytes received: 11
80241006C6C9EE7DD5
Bytes received: 11
80241006C6C9EE7DD5
Bytes received: 11
80241006C6C9EE7DD5
Bytes received: 11
80241006C6C9EE7DD5
Bytes received: 11
802510093DD7E7C3C3F
Bytes received: 11
802510093DD7E7C3C3F
Bytes received: 11
802510093DD7E7C3C3F
Bytes received: 11
802510093DD7E7C3C3F
Bytes received: 11
802510093DD7E7C3C3F
Bytes received: 11
802510093DD7E7C3C3F
Bytes received: 11
802510093DD7E7C3C3F
Bytes received: 11
802510093DD7E7C3C3F
Bytes received: 11
802510093DD7E7C3C3F
Bytes received: 11
802510093DD7E7C3C3F
Bytes received: 11
802510093DD7E7C3C3F
Bytes received: 11
802510093DD7E7C3C3F
Bytes received: 11
802510093DD7E7C3C3F_

moonilightshaker, 2023-08-25:
Output from the "3" button. Pressed 10 times in a row:
_Bytes received: 11
8026100755D7773ED7D
Bytes received: 11
8026100755D7773ED7D
Bytes received: 11
8026100755D7773ED7D
Bytes received: 11
8026100755D7773ED7D
Bytes received: 11
8026100755D7773ED7D
Bytes received: 11
8026100755D7773ED7D
Bytes received: 11
8026100755D7773ED7D
Bytes received: 11
8026100755D7773ED7D
Bytes received: 11
8026100755D7773ED7D
Bytes received: 11
8026100755D7773ED7D
Bytes received: 11
8026100755D7773ED7D
Bytes received: 11
8026100755D7773ED7D
Bytes received: 11
8026100755D7773ED7D
Bytes received: 11
8026100755D7773ED7D
Bytes received: 11
8027100DE7A2FD94E
Bytes received: 11
8027100DE7A2FD94E
Bytes received: 11
8027100DE7A2FD94E
Bytes received: 11
8027100DE7A2FD94E
Bytes received: 11
8027100DE7A2FD94E
Bytes received: 11
8027100DE7A2FD94E
Bytes received: 11
8027100DE7A2FD94E
Bytes received: 11
8027100DE7A2FD94E
Bytes received: 11
8027100DE7A2FD94E
Bytes received: 11
8027100DE7A2FD94E
Bytes received: 11
80281001393E175C3F4
Bytes received: 11
80281001393E175C3F4
Bytes received: 11
80281001393E175C3F4
Bytes received: 11
80281001393E175C3F4
Bytes received: 11
80281001393E175C3F4
Bytes received: 11
80281001393E175C3F4
Bytes received: 11
80281001393E175C3F4
Bytes received: 11
80281001393E175C3F4
Bytes received: 11
80281001393E175C3F4
Bytes received: 11
80281001393E175C3F4
Bytes received: 11
80281001393E175C3F4
Bytes received: 11
8029100EA93323579C1
Bytes received: 11
8029100EA93323579C1
Bytes received: 11
8029100EA93323579C1
Bytes received: 11
8029100EA93323579C1
Bytes received: 11
8029100EA93323579C1
Bytes received: 11
8029100EA93323579C1
Bytes received: 11
8029100EA93323579C1
Bytes received: 11
8029100EA93323579C1
Bytes received: 11
8029100EA93323579C1
Bytes received: 11
8029100EA93323579C1
Bytes received: 11
8029100EA93323579C1
Bytes received: 11
8029100EA93323579C1
Bytes received: 11
802A100F3140621DBD
Bytes received: 11
802A100F3140621DBD
Bytes received: 11
802A100F3140621DBD
Bytes received: 11
802A100F3140621DBD
Bytes received: 11
802A100F3140621DBD
Bytes received: 11
802A100F3140621DBD
Bytes received: 11
802A100F3140621DBD
Bytes received: 11
802A100F3140621DBD
Bytes received: 11
802A100F3140621DBD
Bytes received: 11
802A100F3140621DBD
Bytes received: 11
802A100F3140621DBD
Bytes received: 11
802B1008A845FBCA614
Bytes received: 11
802B1008A845FBCA614
Bytes received: 11
802B1008A845FBCA614
Bytes received: 11
802B1008A845FBCA614
Bytes received: 11
802B1008A845FBCA614
Bytes received: 11
802B1008A845FBCA614
Bytes received: 11
802B1008A845FBCA614
Bytes received: 11
802B1008A845FBCA614
Bytes received: 11
802B1008A845FBCA614
Bytes received: 11
802B1008A845FBCA614
Bytes received: 11
802C10053251078F4BC
Bytes received: 11
802C10053251078F4BC
Bytes received: 11
802C10053251078F4BC
Bytes received: 11
802C10053251078F4BC
Bytes received: 11
802C10053251078F4BC
Bytes received: 11
802C10053251078F4BC
Bytes received: 11
802C10053251078F4BC
Bytes received: 11
802C10053251078F4BC
Bytes received: 11
802C10053251078F4BC
Bytes received: 11
802C10053251078F4BC
Bytes received: 11
802C10053251078F4BC
Bytes received: 11
802C10053251078F4BC
Bytes received: 11
802C10053251078F4BC
Bytes received: 11
802D100681AD35CFA2
Bytes received: 11
802D100681AD35CFA2
Bytes received: 11
802D100681AD35CFA2
Bytes received: 11
802D100681AD35CFA2
Bytes received: 11
802D100681AD35CFA2
Bytes received: 11
802D100681AD35CFA2
Bytes received: 11
802D100681AD35CFA2
Bytes received: 11
802D100681AD35CFA2
Bytes received: 11
802D100681AD35CFA2
Bytes received: 11
802D100681AD35CFA2
Bytes received: 11
802E10031A0F4C026C9
Bytes received: 11
802E10031A0F4C026C9
Bytes received: 11
802E10031A0F4C026C9
Bytes received: 11
802E10031A0F4C026C9
Bytes received: 11
802E10031A0F4C026C9
Bytes received: 11
802E10031A0F4C026C9
Bytes received: 11
802E10031A0F4C026C9
Bytes received: 11
802E10031A0F4C026C9
Bytes received: 11
802E10031A0F4C026C9
Bytes received: 11
802E10031A0F4C026C9
Bytes received: 11
802E10031A0F4C026C9
Bytes received: 11
802F1001F1CAD1715
Bytes received: 11
802F1001F1CAD1715
Bytes received: 11
802F1001F1CAD1715
Bytes received: 11
802F1001F1CAD1715
Bytes received: 11
802F1001F1CAD1715
Bytes received: 11
802F1001F1CAD1715
Bytes received: 11
802F1001F1CAD1715
Bytes received: 11
802F1001F1CAD1715
Bytes received: 11
802F1001F1CAD1715
Bytes received: 11
802F1001F1CAD1715
Bytes received: 11
802F1001F1CAD1715_

ctsims, 2023-08-25:
These are perfect, thanks so much for collecting them! I'll process them and let you know!

ctsims, 2023-08-25:
Internal note: Down/Down + Up/Up from a Phillips wizmote
```
Bytes received: 13
81121100208131D7FE9CA3
Bytes received: 13
81121100208131D7FE9CA3
Bytes received: 13
81121100208131D7FE9CA3
Bytes received: 13
81121100208131D7FE9CA3
Bytes received: 13
81121100208131D7FE9CA3
Bytes received: 13
81121100208131D7FE9CA3
Bytes received: 13
81121100208131D7FE9CA3
Bytes received: 13
8113110020812FD1BABF
Bytes received: 13
8113110020812FD1BABF
Bytes received: 13
8113110020812FD1BABF
Bytes received: 13
8113110020812FD1BABF
Bytes received: 13
8113110020812FD1BABF
Bytes received: 13
8113110020812FD1BABF
Bytes received: 13
8114110020912F7A3C012
Bytes received: 13
8114110020912F7A3C012
Bytes received: 13
8114110020912F7A3C012
Bytes received: 13
8114110020912F7A3C012
Bytes received: 13
8114110020912F7A3C012
Bytes received: 13
8114110020912F7A3C012
Bytes received: 13
811511002091304938FC6E
Bytes received: 13
811511002091304938FC6E
Bytes received: 13
811511002091304938FC6E
Bytes received: 13
811511002091304938FC6E
Bytes received: 13
811511002091304938FC6E
Bytes received: 13
811511002091304938FC6E
```

blazoncek, 2023-08-25:
@ctsims do you have any information on which channels do these packets come?
Perhaps the remote blasts packets one each and every WiFi channel (as it does not know on which channel a receiving device may be connected to WiFi).

ctsims, 2023-08-25:
@blazoncek This is something that the ESP-NOW specs are a little vague on, and sometimes contradictory.

My testing indicates that ESP-NOW uses the current WIFI channel when broadcasting, so in most cases where devices are on the same common network they'll naturally be aligned on transmissions. 

I had the same assumption as you about the message blasting (the fact that there are 11 messages sent above, and 11 wifi channels is definitely a suspicious coincidence), and my [esp remote reference sketch](https://gist.github.com/ctsims/2e292a570222cf17922a6625bf9cfebc#file-wled_esp_now_remote-L75) follows the "blast the message across all channels" approach for simplicity as it's extremely fast and much easier than worrying about aligning the wifi channel.

I'm not sure if the Wizmotes are or are not doing the same thing. As configured by the commercial platform, using the remotes involves onboarding them onto the local wifi network through an app, though, so they do have an opportunity to find a common channel. Also, I'm not sure exactly why the receiver tests would be getting every message if they are indeed limited to the current wifi channel. I *think* that in my initial testing, the remote broadcaster sketch did not work until I forced it to broadcast across all channels, but I didn't document it clearly enough to remember.

It could be that they're just dumping packets because it's cheap and makes it easier for one to get through. ESP-NOW claims to have an ability to confirm the packet was received, but that hasn't gelled with my experience with the protocol. The other thing I thought might be happening on my network was that the Wizmote was broadcasting one message per registered Wiz light on my network, but the numbers never really matched up to confirm that (although I didn't test it carefully).

blazoncek, 2023-08-25:
> My testing indicates that ESP-NOW uses the current WIFI channel when broadcasting, so in most cases where devices are on the same common network they'll naturally be aligned on transmissions.

Indeed it does but that only works if **all** devices are on the same channel which may not be true in multi-AP set-up.
My own testing showed that ESP32 on channel 1 wouldn't deliver a packet to ESP8266 on channel 6.

BTW I encourage you to test and experiment with my `espnow-sync-poc` branch. It will be the future.

TheOnlyHyland, 2023-08-25:
Complete end user stuff here, but got the same remote so hopefully you guys can get a fix in ....wish I could be of help but haven't a clue

blazoncek, 2023-09-05:
@moonilightshaker just received my WiZ remote from amazon.de. It is exactly the same as your images show and works with WLED OOB.
Since I now have a toy I'll make remote user programmable in the same way as JSON IR remote.

blazoncek, 2023-09-27:
@ctsims I finally found the time to look at the messages @moonilightshaker posted.
Oddly enough I cannot find any common field in those regarding button pressed. Did you have any luck?
It may happen that in those messages button is somehow encoded/encrypted in last 6 (?) bytes? BTW the message is not always aligned to 11 bytes, intentional?

blazoncek, 2023-09-27:
Some updates for JSON overrides in 0_15 branch. 3255382132810c865a9c5f9a82f473116e3f8071

zango-me, 2023-11-01:
I had the same problem with my remote which also got imported by REV Ritter GmbH.
I got it from Amazon Germany.

After some hours of trying to get the original firmware of that remote working with WLED (which didn't work) I decided to flash my own (custom) firmware to the remote so it would send the same 13 byte long messages as the other version of the wiz remotes that work with WLED.

Here is my code: [zango-me/wled_wizmote](https://github.com/zango-me/wled_wizmote/)
Theres also a binary under "releases" to flash to the remote directly

SonnyRazzvan, 2023-11-02:
> I had the same problem with my remote which also got imported by REV Ritter GmbH. I got it from Amazon Germany.
>
> After some hours of trying to get the original firmware of that remote working with [...]

Is there any chance you have a wiring diagram for the header? Or a guide for the flashing?

Many thanks in advance!

blazoncek, 2023-11-02:
Please use WLED [forum](https://wled.discourse.group) or Discord for discussion.

pojlFDlxCOvZ4Kg8y1l4, 2023-12-16:
> Okay so I've got a plain ESP32 and tried to flash it with [the binary from this comment](https://github.com/Aircoookie/WLED/issues/3341#issuecomment-1692768451). That gives me invalid header errors: `rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) invalid header: 0x45505f54` I've looked at the [code snippet](https://gist.github.com/ctsims/3522d88029c8d49710cc6686702b9494) from the same comment but that's beyond my understanding of ESP32's flashing and programming - how do I apply that to my specific ESP32? The ESP32 I'm using is [this ESP32-D0WDQ6](https://www.tinytronics.nl/shop/nl/development-boards/microcontroller-boards/met-wi-fi/esp32-d0wdq6-development-board-met-li-ion-li-po-laadcircuit). I don't have any other plain ESP32 available at the moment.

I think I've got it figured out now, see attached txt file
[amazon rev wiz remote esp32.txt](https://github.com/Aircoookie/WLED/files/13694235/amazon.rev.wiz.remote.esp32.txt)

pojlFDlxCOvZ4Kg8y1l4, 2023-12-22:
The REV Wiz remotes I've got are not working in the current beta, build and flashed sometime early this week.

blazoncek, 2023-12-22:
Just figure out button codes and use JSON file. Same as JSON IR remotes.

spectrenoir06, 2024-01-10:
I have a dump of the new firmware who send 11 unidentified bytes and not 13 byte we know from the previous version:
[dump.zip](https://github.com/Aircoookie/WLED/files/13895293/dump.zip)
If somebody is good at reverse engineering

spectrenoir06, 2024-01-10:
I also have some dump of the espnow pqt:
for different buttons

1:
80 2f 01 00 00 e5 6d 1c 43 87 72 
80 30 01 00 00 66 ec d8 9d 6b 41 
80 31 01 00 00 9b 84 14 2f d4 ab 
80 32 01 00 00 69 34 ab a2 72 f3 
80 33 01 00 00 aa 51 ca dc ff e4 
80 34 01 00 00 2a 1d e1 70 56 87 
80 35 01 00 00 23 71 6f bc 9d dc 
80 36 01 00 00 28 28 90 1b 03 67 
80 37 01 00 00 3f 45 19 10 a8 08 
80 38 01 00 00 d3 f4 c0 06 db 3e 
80 39 01 00 00 48 3b e7 62 51 0d 
80 3a 01 00 00 b4 92 33 95 68 18 
80 3b 01 00 00 59 de d5 1c ee a6 
80 3c 01 00 00 f2 34 53 4e 24 6e 
80 3d 01 00 00 77 68 01 e9 cd 38 
80 3e 01 00 00 bb 89 1d e4 b5 89 
80 3f 01 00 00 c5 c2 70 5c d2 30 
80 40 01 00 00 22 53 ac 2a e1 c2 
80 41 01 00 00 1e e6 89 5f d8 c4 
80 42 01 00 00 08 a8 c2 69 31 b4 
80 43 01 00 00 e2 af d1 b1 27 54 
80 44 01 00 00 f7 70 fa 17 91 fd 
80 45 01 00 00 0b e6 26 a1 4a 54 
80 46 01 00 00 36 da ac 9f df 98 
80 47 01 00 00 70 c4 00 50 bb 14 
80 48 01 00 00 f5 4d 3d ad b9 78 
80 49 01 00 00 42 29 5e 25 7c b5 
80 4a 01 00 00 3c 49 57 5a 8c 23 
80 4b 01 00 00 c2 7f 2a d1 56 eb 
80 4c 01 00 00 fb 24 ac 46 a1 bd 

2
80 4d 01 00 00 cd ab 0f 6e 8f 77 
80 4e 01 00 00 57 ab 26 41 64 1b 
80 4f 01 00 00 7b 24 55 c6 87 ff 
80 50 01 00 00 f3 5c 92 8d 9e f8 
80 51 01 00 00 d2 3f 84 b3 25 62 
80 52 01 00 00 47 5e f6 82 93 7b 
80 53 01 00 00 c9 f7 72 a7 af c2 
80 54 01 00 00 b1 27 0a 97 b9 64 
80 55 01 00 00 9d 9e 7a 70 72 6c 
80 56 01 00 00 fe 68 91 59 80 de 
80 57 01 00 00 21 08 15 53 28 14 
80 58 01 00 00 83 14 00 ef 20 1d 
80 59 01 00 00 e3 ac 36 86 0e 86 
80 5a 01 00 00 28 b7 d1 a8 36 0b 

3
80 4d 01 00 00 cd ab 0f 6e 8f 77 
80 4e 01 00 00 57 ab 26 41 64 1b 
80 4f 01 00 00 7b 24 55 c6 87 ff 
80 50 01 00 00 f3 5c 92 8d 9e f8 
80 51 01 00 00 d2 3f 84 b3 25 62 
80 52 01 00 00 47 5e f6 82 93 7b 
80 53 01 00 00 c9 f7 72 a7 af c2 
80 54 01 00 00 b1 27 0a 97 b9 64 
80 55 01 00 00 9d 9e 7a 70 72 6c 
80 56 01 00 00 fe 68 91 59 80 de 
80 57 01 00 00 21 08 15 53 28 14 
80 58 01 00 00 83 14 00 ef 20 1d 
80 59 01 00 00 e3 ac 36 86 0e 86 
80 5a 01 00 00 28 b7 d1 a8 36 0b

Chris8-1, 2024-03-16:
> Is there any chance you have a wiring diagram for the header? Or a guide for the flashing?
>
> Many thanks in advance!

I got it done with the bin file from zango-me. (Thank you for that)
https://wled.discourse.group/t/how-do-i-flash-a-wizmote/11326