wled/WLED #3342: 2D Matrix effects

dksalpha, 2023-08-23:
### What happened?

Some 2D matrix effects are not working properly. Also unable to load pixel art from presets after saving.

### To Reproduce Bug

I have 3 segments (8 pix, 24 pix, 64 pix) in total, Segments 1 and 2 connected to GPIO2, 3rd segment is a sole 8x8 Matrix connected via GPIO1 in an ESP8266. Configured in settings as 8x12 Matrix of 3 panels (8x1, 8x3, 8x8). 

### Expected Behavior

Effects such as Akemi and Scrolling Text work perfectly, but other effects such as Blackhole, Coloured Bursts and Plasma Ball are not. No noticeable issue if running 0.14.0-b1. 

Also when sending pixel art via browser using WLED-PixelArtConverter, images load perfectly. However after saving to a preset pixel image does not load to matrix correctly if at all. For reference, Scrolling text as a preset works fine. Not tested in 0.14.0-b1. 

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b4

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2023-08-23:
Matrix effect will not work if you are on low brightness and Use global LED buffer is not selected . 
Look here for demo https://discord.com/channels/473448917040758787/779395228624617512/1139552308112412863

You have added pixel art to this post and not sure why but   try the other tool 
https://github.com/ajotanc/PixelMagicTool

dosipod, 2023-08-23:
For effects try to  enable Use global LED buffer  and test again 
If you are trying to mix 2d and 1d setup then adding the strip as a matrix is a workaround but you can mix the setup if you 
add the strip to the end of matrix 
[image]
For pixel art try this tool https://github.com/ajotanc/PixelMagicTool

If any of above does not work then might be and  post video or better on discord

dksalpha, 2023-08-23:
Thanks for the advise. Mixed setup is fine, but definitely a user error somewhere as working perfectly after taking a step back. As Text and Akemi were working fine, in my confusion I made the assumption all settings were fine as in the past I've used those as the test effects.
Excuses aside, my apologies!!!

Re pixel art, still having preset issue, and only used it because of Knowledge base. Will definitely use Pixel Magic going forward as its working as advertised. Thanks for pointing me towards that tool.

closed 2023-08-23 by dksalpha