wled/WLED #3343: Multiple Animated Pixel Art SHUFFLE

dksalpha, 2023-08-23:
As I understand it, to make animated pixel art a user would save each frame as a preset and then run as playlist. If using multiple animated art you wouldn't be able to use the shuffle feature without ruining animations.

Is it possible to add a function to presets to create a playlist from other playlists, or another solution? This would allow Animated Pixel Art (also just regular effects on a flat strip/matrix) to appear in a loop or shuffle at users discretion, instead of as a loop only.

blazoncek, 2023-09-28:
Nested playlists are currently impossible and may not be possible in the future with current approach.
However there is a GIF player in the works.

Liliputech, 2025-08-24:
https://github.com/wled/WLED/pull/3835