wled/WLED #3344: Multi-segment presets not syncing properly between controllers

CaptainUns, 2023-08-24:
### What happened?

I created a multi segment preset on my main controller of a three solid color repeating pattern. I have sync activated on the main controller and the main and slave controllers on the same sync group. I also have both controllers set to receive brightness, color, effects, segment options, and bounds. When I activate the preset on the main controller, the slave controller will only display the grouping and spacing from the first segment and the color from the second segment, but no other colors or segment options

### To Reproduce Bug

Preset is 3 segments:
First segment: Start LED 0, Grouping 2, spacing 4, solid red
Second segment : Start LED 2, Grouping 1, spacing 2, solid warm white
Third segment: Start LED 3, Grouping 2, spacing 4, solid green
All three segments are set to the full length of the LED run. Pattern should be RRWGGWRR…repeating to the end of the run
Both controllers are set to the same sync group.
Sync options for both controllers are set to receive brightness, color, effects, segments and bounds.
Sync is turned on on the main controller (the controller with the preset in question).
When the preset is activated the slave controller does not properly sync/display the main controller preset

### Expected Behavior

The slave controller should automatically create three duplicate segments, identical to the three segments on the main controller and display the identical pattern as the main controller

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 14.0-b4 (2308110)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-08-25:
Do you have all segments, that you want to sync, selected?
Non-selected segments do not sync.

CaptainUns, 2023-08-25:
> Do you have all segments, that you want to sync, selected?
> 
> Non-selected segments do not sync.

Yes, all segments I am attempting to sync are selected. The segments are saved as a preset with all three selected. Note that for some reason the spacing from segment 1 (but not color) and the color from segment 2 (but not spacing) are what ends up syncing

CaptainUns, 2023-08-25:
If I manually create the matching segments on the receiving unit then the colors/pattern from the sending unit sync correctly, but from what I understand, the receiving unit should be automatically creating/removing/adjusting segments to match the sending unit, which is what is not happening. If the segments on the receiving unit are created manually and then the preset on the sending unit is changed to a single segment effect, all three segments remain on the receiving unit and only the first segment receives the new effect without matching the new segment options & bounds.

blazoncek, 2023-08-25:
What you are describing is correct and expected behaviour.

Receiving device should have segments set up (and selected) and then those segments will receive data from sending device (including their length if bounds is checked).

Unfortunately what you *expect* is not implemented since nonexisting segments are not synced to receiving device.

I.e. if master has 2 segments in 1st preset and 1 segment in another preset, the slave (receiver) will not get information that a segment was removed when switching from preset 1 to preset 2. This is current implementation and I am afraid it will be difficult to proprly implement it otherwise.

CaptainUns, 2023-08-28:
I may be missing something in the explanation, but from what I can see in this video, he is able to accomplish exactly what I am describing.

https://youtu.be/Yd-2Sl8v5FM?si=WqDSQoVv4s7GdhVh

At 4:41 he selects a 2-segment preset from the master device sending the signal, and the slave device receiving the signal creates a second segment to match the preset selected. Prior to this, the slave device had only 1 segment created and automatically created a second one matching all the settings of the master device.

Given what you said above, is there another setting I'm not aware of or something else I'm missing to get this to work?

softhack007, 2023-08-28:
> but from what I can see in this video,

As a general fact of life - don't trust in what people show on youtube. Consult our KB if you need information https://kno.wled.ge/

> Given what you said above, is there another setting I'm not aware of or something else I'm missing to get this to work?

That's a good question.

CaptainUns, 2023-08-28:
> As a general fact of life - don't trust in what people show on youtube. 

Cool, cool - not sure how talking down to me like I'm a child helps anyone in this situation, but thanks. Super helpful. Is there another "contributor" who can maybe provide actual assistance here?

Your intended point here (and the fact that I have checked the KB multiple times) aside, are you suggesting that this guy created a video with the intention of explaining how the sync function works, and (from what I can tell) explained everything correctly except this one feature, which he specifically faked for the purposes of...?)

I approached Discord first and it was recommended (by another dev, actually) that I open a bug report on this, which is why I came here.

> That's a good question.

Again, a super helpful and specific response.

He accomplished this somehow, and if you know how, wouldn't it save everyone time for you to either point me to the specific place on the KB that I must have missed or just actually tell me?

In general I've always found the WLED community helpful,welcoming and accommodating, so the undertone here of "figure it out for yourself" is surprising.

blazoncek, 2023-08-29:
@CaptainUns there is a fundamental difference between 0.13 (the video is using) and 0.14 (you are using) on how segments operate.
This difference is needed to allow seamless color, palette and effect (yes, almost ready to be published) transitions (or blending). Due to this change the segment syncing works a bit different between 0.13 and 0.14 and due to technical reasons may never be the same.

That being said it is still valid to open a bug report but unfortunately it may take a while until we find a solution. If at all.

softhack007, 2023-08-29:
> In general I've always found the WLED community helpful, welcoming and accommodating, so the undertone here of "figure it out for yourself" is surprising.

Hi, I'm sorry for my short response - it was late (around 2am) and I had a long day - I did not want to hurt your feelings.

I was confused what the real problem is, as you just gave a YT video as reference of "expected behaviour". Maybe the fact I'm not a native speaker has added some misunderstanding on my side.
Thanks @blazoncek for the technical answer.

Please accept my apology, as I shouldn't write resposes when my brains are already sleeping ;-)

blazoncek, 2023-08-29:
I managed to get something going on.
If you are willing to compile and alpha-test please use `espnow-sync-poc` branch.

CaptainUns, 2023-08-29:
@blazoncek Awesome! I will try tonight when I get home.

CaptainUns, 2023-08-30:
Compiled and uploaded successfully, but Sync does not appear to be working - on/off, brightness, effect, segments/bounds, none of them are responsive. Confirmed both master and slave are on same sync group, sync turned on for master device.

blazoncek, 2023-08-30:
[this one](https://github.com/Aircoookie/WLED/commit/c95ea7bbe5a4a8c3f5c7d06e875d31f16ea91da1) should work better

CaptainUns, 2023-08-30:
@blazoncek unfortunately same result - successful compile/upload but sync still not working

blazoncek, 2023-08-31:
my own tests show that it is working. segments are created and removed, effects and colors also synced.

CaptainUns, 2023-08-31:
@blazoncek confirmed both master and slave devices have installed build 2308300, master has sync turned on and both are on the same sync groups, but still no luck with anything syncing.

What's strange is if I go back to latest release build (2308110) on the master device, sync works again AND segments now sync as expected to slave device. That's with your build on slave device and release build on master device. No change in sync settings between builds. But for whatever reason with your build on both devices sync does not appear to work.

Don't know board type make a difference here? I'm using esp8266 boards (d1 mini).

CaptainUns, 2023-09-04:
@blazoncek Just following up here - are you using ESP32 or ESP8266? Would that potentially make a difference?

blazoncek, 2023-09-04:
No, as I am testing cross device. I have them sync w/o issues.
8266 does not support as many segments as 32 does, that's the main difference.

Still, as mentioned earlier, segment handling in 0.14 is completely rewritten from scratch and incompatible with 0.13. What I did was add a *band-aid* to try to simulate old behaviour it may not behave *exactly* as 0.13 did.

If you want to rely on segment bounds syncing then, I'm afraid, you will need to use 0.13 until some better idea comes around. That is also why I'm not closing this issue.

blazoncek, 2023-09-18:
@CaptainUns please try 0_15 branch and report back.

CaptainUns, 2023-09-19:
Installed 0_15 on both master and slave devices, sync does not appear to be working. Double checked both devices on same sync channel, sync button "ON" for master device, no luck. Also have a few other devices on the same sync receive channel that are on older release versions, and they aren't picking up any sync signal from master device either.

blazoncek, 2023-09-20:
My own testing show sync working so I will assume your setup to be misconfigured.
Unless you post your configuration settings (cfg.json) from master and slave devices, I'm afraid I cannot help further.

WizBangCrash, 2023-12-01:
@blazoncek I have been having the same issue with 0_14 so downloaded and compiled 0_15 this evening to try out your fix, but I am having the same issue as @CaptainUns i.e. it does not appear to be working.

I have 3 WLED controllers. Two of them have now been updated to the HEAD of 0_15 branch and one has been left on the latest 0.14 release.
One 0_15 controller is configured to receive sync on group 2 & 3
The 0_14 controller is sending sync on group 2. The other 0_15 controller is sending sync on group 3
Nothing seems to work when the 0_15 controller is sending sync, but when the 0_14 controller sends sync updates the receiving 0_15 controller changes. Just with missing the extra segment info.

WizBangCrash, 2023-12-02:
I made a bit more progress with testing this today.
Updated all WLED controllers to latest 0_15 branch.
Realised I had to set the `Send notifications on direct change` option in the Sync configuration, so that was why nothing was being sent from my other controllers 🤦

I now have all my controllers syncing again, but only if they have the same segment setup.
If I sync a preset that has multiple segments to a controller that only has one segment then I get a `Failure: error 9' appear on the screen

I switched to using the JSON API with known good API requests and get the same behaviour. A single segment API call works fine. A multi-segment API call responds with `400 Bad Request` and the following JSON response:
`{
  "error": 9
}`

This is the JSON API request content:
`{
  "on": true,
  "bri": 39,
  "transition": 7,
  "mainseg": 0,
  "seg": [{
    "id": 0,
    "start": 0,
    "stop": 206,
    "grp": 1,
    "spc": 20,
    "of": 0,
    "on": true,
    "frz": false,
    "bri": 255,
    "cct": 127,
    "set": 0,
    "n": "Red",
    "col": [
      [255, 0, 47],
      [0, 0, 0],
      [0, 0, 0]
    ],
    "fx": 0,
    "sx": 128,
    "ix": 128,
    "pal": 0,
    "c1": 128,
    "c2": 128,
    "c3": 16,
    "sel": true,
    "rev": false,
    "mi": false,
    "o1": false,
    "o2": false,
    "o3": false,
    "si": 0,
    "m12": 0
  }, {
    "id": 1,
    "start": 4,
    "stop": 206,
    "grp": 1,
    "spc": 20,
    "of": 0,
    "on": true,
    "frz": false,
    "bri": 255,
    "cct": 127,
    "set": 0,
    "n": "Green",
    "col": [
      [12, 255, 8],
      [0, 0, 0],
      [0, 0, 0]
    ],
    "fx": 0,
    "sx": 128,
    "ix": 128,
    "pal": 0,
    "c1": 128,
    "c2": 128,
    "c3": 16,
    "sel": true,
    "rev": false,
    "mi": false,
    "o1": false,
    "o2": false,
    "o3": false,
    "si": 0,
    "m12": 0
  }, {
    "id": 2,
    "start": 8,
    "stop": 206,
    "grp": 1,
    "spc": 20,
    "of": 0,
    "on": true,
    "frz": false,
    "bri": 255,
    "cct": 127,
    "set": 0,
    "n": "Orange",
    "col": [
      [255, 119, 15],
      [0, 0, 0],
      [0, 0, 0]
    ],
    "fx": 0,
    "sx": 128,
    "ix": 128,
    "pal": 0,
    "c1": 128,
    "c2": 128,
    "c3": 16,
    "sel": true,
    "rev": false,
    "mi": false,
    "o1": false,
    "o2": false,
    "o3": false,
    "si": 0,
    "m12": 0
  }, {
    "id": 3,
    "start": 12,
    "stop": 206,
    "grp": 1,
    "spc": 20,
    "of": 0,
    "on": true,
    "frz": false,
    "bri": 255,
    "cct": 127,
    "set": 0,
    "n": "Blue",
    "col": [
      [0, 89, 255],
      [0, 0, 0],
      [0, 0, 0]
    ],
    "fx": 0,
    "sx": 128,
    "ix": 128,
    "pal": 0,
    "c1": 128,
    "c2": 128,
    "c3": 16,
    "sel": true,
    "rev": false,
    "mi": false,
    "o1": false,
    "o2": false,
    "o3": false,
    "si": 0,
    "m12": 0
  }, {
    "id": 4,
    "start": 16,
    "stop": 206,
    "grp": 1,
    "spc": 20,
    "of": 0,
    "on": true,
    "frz": false,
    "bri": 255,
    "cct": 127,
    "set": 0,
    "n": "Yellow",
    "col": [
      [255, 234, 46],
      [0, 0, 0],
      [0, 0, 0]
    ],
    "fx": 0,
    "sx": 128,
    "ix": 128,
    "pal": 0,
    "c1": 128,
    "c2": 128,
    "c3": 16,
    "sel": true,
    "rev": false,
    "mi": false,
    "o1": false,
    "o2": false,
    "o3": false,
    "si": 0,
    "m12": 0
  }]
}`

WizBangCrash, 2023-12-02:
Cutting and pasting the above JSON into a preset works fine, but the API call gets an error `9`??
Even after cutting and pasting and proving the preset works fine on that specific controller I was unable to get anything other than the first segment synced to other controllers.

I tried creating 5 segments with the same offset on another controller and then syncing it but, again only the first segment gets synced.

3rdStng, 2023-12-03:
Same issue on my two digquad's running WLED 0.14.0 (build 2310130). Main unit has the Sync option enabled and will sync colors, offsets, groupings, etc to the secondary unit. However with a preset that has multiple segments, only the grouping, spacing, offset information is being sent to the second controller. The second controller remains to have segment 0 defined

3rdStng, 2023-12-03:
I flashed back to WLED 0.13.3 (build 2208222) and the segments are sent over perfectly
[image]
[image]
[image]
[image]

3rdStng, 2023-12-03:
Thank you @WizBangCrash. I knew there was a difference in syncing between the two. I also saw a few updates (14.1 and 15) in the thread too, but nothing concrete for the solution. Just reporting that reverting back to 13.3 works for now. Happy to test on future builds, if needed.

blazoncek, 2023-12-04:
@WizBangCrash @3rdStng @CaptainUns  please read my last post above and provide what I requested. Otherwise there is no need to post in this issue.

I am willing to provide (over on Discord) a debug binary which may help trace down the root cause (as I cannot reproduce). It will require from you to upload it to both devices: master and slave and do serial capture on both.
I may also require from you do do network trace using Wireshark or similar tool.

WizBangCrash, 2023-12-04:
Hi @blazoncek  I made a bunch of progress using the NET Debug feature last night ( I have no capability to do serial capture on any of the my controllers as they are in totally inaccessible places while my Christmas lights are up).
I did managed to get the segment syncing working with 0.15. Just not the way I was expecting it to work.

I am still a little confused over what is "Segment Options" and what is "bounds". I thought I only needed to sync the options as all my strips are different lengths and so the bounds do not necessarily apply. It turns out segments only get sync'd if the receive "bounds" option is enabled.

I am now not sure I can do what I want to do. I have a 5 segment preset on the master controller that I want to sync with the other controllers. 
[xmas-tree-lights-preset.json](https://github.com/Aircoookie/WLED/files/13549739/xmas-tree-lights-preset.json)

The master controller has 107 LEDs, but my other strips range from 196 to 344 LEDs. When I sync the preset it replicates the pattern but sets them all to a maximum of 107 LEDs. Leaving the rest of the strip dark.
I thought I could just sync the segment options to overcome this, but then the actual segments do not get created.

Here are the configs for the master and one of the slaves:
[wled_cfg_Athom-5-Office-ESL (2).json](https://github.com/Aircoookie/WLED/files/13549800/wled_cfg_Athom-5-Office-ESL.2.json)
[wled_cfg_Athom-1-Garage.json](https://github.com/Aircoookie/WLED/files/13549801/wled_cfg_Athom-1-Garage.json)

blazoncek, 2023-12-04:
Bounds = start & stop positions
Options = mirroring, grouping, spacing, etc

You would want bounds syncing when you would want *all devices* to *completely mirror* master semgment sizes.
I.e. If master displays 3 segments so should slave.

When you only sync segment options slave segments (number & dimensions) remain unchanged.

blazoncek, 2023-12-04:
> I am now not sure I can do what I want to do. I have a 5 segment preset on the master controller that I want to sync with the other controllers.
> [xmas-tree-lights-preset.json](https://github.com/Aircoookie/WLED/files/13549739/xmas-tree-lights-preset.json)

A bit unorthodox IMO. A similar output cam be achieved with single segment, Colorful effect and C9 palette. But that discussion is for Discord.

WizBangCrash, 2023-12-04:
@blazoncek  Thanks for the updates. Much appreciated.
I think I've managed to get it sorted tonight. I've made the controller with the largest number of LEDs the master (syncing bounds and options) and when it syncs to the slaves with the smaller LED counts the segment count is automatically reduced to the maximum number of pixels each slave has attached. Basically using a side effect of the code, but works well for me :-)

I'll stop posting to the issue now leave any other conversations about segment set up to Discord.
I am happy to confirm that the 0.15 branch works as expected though :-)

CaptainUns, 2023-12-04:
@WizBangCrash Sorry to hear you had this same issue but glad to hear it wasn't just me, and also that you found a fix!

After debug, what did you change or do differently to get segments to sync on 0.15?

CaptainUns, 2023-12-05:
@blazoncek So as an example, what I was trying to do was create a preset with 3 segments that would sync from the master device to multiple slave devices. I was trying to create a holiday pattern of 2 red, 1 white, and 2 green, 1 white, etc. by creating 3 segments:

1. Grouping 2, spacing 4, starting led 0 (red)
2. Grouping 1, spacing 2, starting led 2 (white)
3. Grouping 2, spacing 4, starting led 3 (green)

Based on what you are saying above, this should still work but bounds must be utilized and so all devices will mirror the master device, including the total number of lit LEDs? If that is the case and I am understanding correctly, that is not what I am experiencing. For whatever reason, with segment options and bounds selected to sync on all devices, I only get 2 white LEDs lit with 4 LED spacing on all slave devices. It would make more sense to me if it was 2 red LEDs and not white, because that would mean only the first segment is syncing, but this is somehow combining the grouping of segment 1 and the color of segment 2 and sending that as a single segment to all slave devices, rather than creating 3 segments with the same settings as my master device.

WizBangCrash, 2023-12-05:
Segments definitely sync correctly in the 0.15 branch. 
It was my understanding of how the various options affect sync that wasn't clear.
I have to sync `bounds` as I have an odd segment setup (look at the xmas-tree-lights-preset.json in previous post), but I am able to get around this by using the controller with the longest run of LEDs as the master. When this syncs to the slave controllers they truncate/reduce the `stop` field to their maximum LED count and everything works as expected i.e. All segments are created and the `options` are set correctly for each, but the `stop` value is reduced. :-)

...and yes @blazoncek I understand that 0.15 may change a lot before release, but I'm just trying to get this year's Christmas lights working so I am happy to build my own binaries for now :-)  Next year will be a new challenge, I am sure,  as I plan to add a lot more lights, but that's all part of the fun...

blazoncek, 2023-12-05:
Do not get me wrong. All of my lights run 0.15, including outdoor display. So it is pretty stable but if you do find a bug, let me know on Discord.

blazoncek, 2023-12-31:
I am closing as numerous tests showed segment syncing as designed. If the user expects different kind of synchronisation it may be out of WLED scope and perhaps an external software may need to be used.

closed 2023-12-31 by blazoncek