wled/WLED #3347: issue loading your presets

JevVan, 2023-08-26:
### What happened?

I was changing up my presets and at some point it failed to load and gave me the issue loading presets error.

After some googling and a bunch of troubleshooting with backup and restore I think I got it to a repeatable situation.  It probably has something to do with my switching the leds from a couple strands on pins 2 & 4 to a matrix connected only to pin 2.  

### To Reproduce Bug

Using the following .json backup of the presets, if you delete the one in slot 55, then press Config, then press Back,  You get the error, "Failed to load presets"

Looking a the presets file, it is add a bunch of spacing in the .json when deleting the preset in spot 55.  Running it through a cleaner after removing number 55 the same file will work.  So I believe it is just a bunch of extra space.

I was thinking it was only happening when deleting the last preset, but I cannot make it occur again with new presets even if I put them in slot 55.  So, in the end, I'm not sure how I got here unless it's the change in configuration.

Here is the preset that will break when 55 is deleted:
[wled_presets_WorkingWith55.json.txt](https://github.com/Aircoookie/WLED/files/12446678/wled_presets_WorkingWith55.json.txt)

And here is a backup of the broken one after deleting item in slot 55
[wled_presets_BrokenWithout55.json.txt](https://github.com/Aircoookie/WLED/files/12446676/wled_presets_BrokenWithout55.json.txt)

### Expected Behavior

It just works!

### Install Method

Binary from WLED.me

### What version of WLED?

WLED version 0.14.0-b4

### Which microcontroller/board are you seeing the problem on?

ESP32-S2

### Anything else?

Not 100% sure on the controller.  The stick on back says NODEMCU     ESP32S      ver 1.1

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-08-27:
Duplicate of #3181 #2886 #3139 #2793 #2755 #1981 #2434 and possibly others.

A known bug with a few workarounds and remedies in place but it can still happen as root cause is still undetermined.

closed 2023-08-27 by blazoncek

Phobos-7, 2023-11-22:
Unfortunately, this issue wasn't fixed in the latest 0.14.0.
I am not using an Android app, only a web interface.
I got "Sorry, there was an issue loading your presets!" error right after I backed up my presets to a json file...

JevVan, 2023-11-23:
JSON file gets corrupted I was able to deal with it by opening manually
finding the messed up parts and fixing it.

blazoncek, 2025-12-29:
Anyone stumbling on this: Presets on ESP32 get corrupted (99.999%) on save by incompatible flash mode used when flashing ESP for the first time.
To resolve the issue *you will need to flash new bootloader using DIO or DOUT mode*.