wled/WLED #3353: Mqtt Broker limition of 32 character

dosipod, 2023-09-01:
### What happened?

As per the original post in discord  ( we are only posting as asked  ) we have tested  with some of the free sites  and  Mqtt Brokers larger  then 32 character  will not be allowed in  wled  sync page 

Original post :
https://discord.com/channels/473448917040758787/473449548279185408/1146929775798136842 
Broker tested is HiveMq which works fine with other mqtt clients 

### To Reproduce Bug

Enter  more then 32 char in Mqtt Broker address 

### Expected Behavior

More then 32 should be allowed  ( the site we tested needed 51 char but could be more in the future )

### Install Method

Self-Compiled

### What version of WLED?

14.0.b4

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

closed 2023-09-01 by Aircoookie

dosipod, 2023-09-02:
Why the rush to close , could at least to test but i guess

blazoncek, 2023-09-02:
It was closed automatically as it was linked to PR.
You can always re-open if you wish.

dosipod, 2023-09-02:
No worries  , i am just not sure how to test it  which i wanted to  add result here or to the [PR](https://github.com/Aircoookie/WLED/pull/3354)  , there is a compile flag now ?

blazoncek, 2023-09-02:
If you don't disable MQTT then add -D MQTT_MAX_SERVER_LEN=64 and if topic length is giving you griefs then -D MQTT_MAX_TOPIC_LEN=128 will help you out.

dosipod, 2023-09-03:
Sure you mean help the guy who uses that free site   site :)  as i am only testing to help you out debug this I did try 54 before but not  64 or  MQTT_MAX_TOPIC_LEN=128
[image]