wled/WLED #3355: Preset brighness doesn't get applied if a brighness change from a previous command is still executed

KlausHans, 2023-09-04:
### What happened?

I have a potentiometer (WH138-1) with a build in switch directly connected to Wled. The switch is triggered if the potentiometer is turned to the maximum counter clockwise position ("light off"). The switch triggers an off preset.
In addition i have a presence detector (Knx, connected to Wled via Home Assistant) that also has to switch Wled.
The problem is, when the potentiometer is used and the light gets switched off by the potentiometer, the brightness is at a minimum. Now, if the presence detector switches Wled on, the brightness is still at a minimum. No problem, i thought, and added a brightness value to the off-preset that gets triggered by the potentiometer switch.

Here is where it gets curious and what my issue is: if i turn down the potentiometer in one go to the off-switch position, the brightness still turns down when the off-preset with brightness value is triggered by the switch. Then the brightness i want for the presence detector is not applied. If i turn the potentiometer down and stopp before the off-switch position, wait a moment and then turn to the off position, the brightness gets applied.
I tried to increase the transition time for the off preset, but that didn't helped so far.
This seems to be some sort of race condition.

### To Reproduce Bug

- have a potentiometer with integrated switch 
- configure the potentiometer as analog button for brightness control 
- configure a preset with a brightness (e.g. 128) that gets triggered if the switch of the potentiometer is triggered in the off position
- turn the potentiometer down to the switch position in one go so the preset is triggered
-> the brightness of the preset isn't applied

### Expected Behavior

- the preset brightness gets applied
(in other words: commands/preset changes are executed strictly sequentially)

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b4

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

The issue is not that important, i just wanted it to be documented. Maybe it is even expected behaviour due to technical reasons?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-09-04:
Please post API for all presets used.

KlausHans, 2023-09-04:
Sure.

Preset "On":
`{"on":true}`
[image]

Preset "Off":
`{"on":false,"bri":127,"transition":20}`
[image]

Some additional screenshots:
[image]
[image]

blazoncek, 2023-09-04:
What about HA and/or Knx?

When you use potentiometer as a brightness "driver" its position determines absolute brightness level. So when you turn potentiometer all the way down and then trigger a switch change, you effectively do 2 things: set brightness to 0 (using potentiometer; it also sets "previous" brightness to 1) and toggle WLED off (using switch; even though it was already off by potentiometer). If you turn potentiometer quickly you may get "previous" brightness >1 as potentiometer is sampled only a few times per second.
When you try to turn WLED on using external device, your potentiometer is still at lowest value which will set brightness to 1 (or whatever "previous" brightness was).

This is not a bug, it is a design choice.

KlausHans, 2023-09-05:
Thank you for your time. I've read your answer 5 times and i think now i've got it.

I didn't knew that "global brightness" also switches Wled off in the lowest position. But that doesn't matter.

> If you turn potentiometer quickly you may get "previous" brightness >1 as potentiometer is sampled only a few times per second.

Ok, that seems to be the core of my issue. Wled reads a potentiometer a change after the switch was triggered because of the low sample rate, even though the real last change of the potentiometer was before the switch was triggered. Correct?

So without a high potentiometer sample rate i have no chance to "fix" it inside of Wled.

blazoncek, 2023-09-05:
No. In fact there is no reason to have "Off" switch with potentiometer except if you add an "offset" resistor to keep voltage above 0V when the potentiometer is all the way down.

closed 2023-09-05 by KlausHans