wled/WLED #3357: Adding more than 3 data outputs

Doyle4, 2023-09-05:
### What happened?

4th data line doesn't have same options as other 3

### To Reproduce Bug

try and add 4 data lines

### Expected Behavior

4 data lines

### Install Method

Binary from WLED.me

### What version of WLED?

0.14-b4

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Anything else?

Sorry not filled in the bits, I was wondering if its possible to add more than 3 data lines? when pressing + after 3rd the 4th one isn't the same as the other 3, is it possible to have 4? Thanks.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-09-05:
That is a hardware limitation of C3.

closed 2023-09-05 by blazoncek

Doyle4, 2023-09-05:
Thanks for getting back to me, which boards could use upto 4? Thanks

blazoncek, 2023-09-05:
Please use [WLED forum](https://wled.discourse.group) or Discord for help and support questions.