wled/WLED #3359: Playlist: can not set the duration to under 1 second

KlausHans, 2023-09-06:
### What happened?

Based on https://github.com/Aircoookie/WLED/issues/346#issuecomment-1219882212 i tried to set a duration for a playlist entry to 0.1 s. But the minimum duration i can set in the gui is 1 s. If i change to json view and set "dur":[1] (0.1 s) the value doesn't get saved. Instead the default value of 100 (10 s) is saved.
Tested Firefox and Chromium.

### To Reproduce Bug

- create preset
- add the preset in a playlist
- try to set the duration of the added preset to <1 s (e.g. 0.1 s)

Some screenshots of my setup:
Preset:
[image]

Playlist with fresh input of "dur":[1]:
[image]
after clicking save and open the playlist again;
[image]

### Expected Behavior

I can set a duration lower than 1 second.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b4

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-09-06:
Minimum duration value is 0.2s. This is a design choice.
Tested on multiple units.

KlausHans, 2023-09-06:
0.2 works indeed. Thank you for clarifying. This info could be added to the documentation.

KlausHans, 2023-09-06:
It was both. A min constraint on the input in the Ui is a good idea.

blazoncek, 2023-09-06:
> @blazoncek not sure if this was about a value entered in a field in web UI, or just direct json in presets? If it's about web UI, maybe we could add a "min" to the input field?

UI already has the "min" property and a "step" property (min="0.2" step="0.1"). This is converted to 1/10ths of a second and submitted in JSON.

KlausHans, 2023-09-06:
Ok but it doesn't show any hint or error if i put 0.1 in.

blazoncek, 2023-09-06:
> Ok but it doesn't show any hint or error if i put 0.1 in.

True, I noticed that. There is a validation, but does not work as expected.