wled/WLED #3362: OPNSense no uPnP intermittent Wifi connection

Apprisco, 2023-09-08:
### What happened?

Without uPnP (such as on OPNSense Router), the web gui will take forever to load, and there will be very intermittent wifi connections for certain WLED devices.

### To Reproduce Bug

Disable uPnP in router.
Observe new WLED devices have very intermittent web-gui and app access.

### Expected Behavior

As wLED is LAN based, it should only rely on DHCP or static IP.
uPnP should not be required for WLED, and thus without uPnP, such as for homelab use, WLED should not have intermittent connections.

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.3 & 0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
No log output at all.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-09-08:
WLED does not use uPnP.
The problem might be in your network set-up.

softhack007, 2023-09-08:
@Apprisco is this the same problem as in your new report 
* https://github.com/Aircoookie/WLED/issues/3363

If yes, should we close this one in favour of the new report?

softhack007, 2023-09-09:
duplicate of  #3363

closed 2023-09-09 by softhack007