wled/WLED #3363: Endless reconnect to wifi network, not much information in debug build.

Apprisco, 2023-09-08:
### What happened?

[image]

Seems like i'm getting constantly disconnected to the point where I can barely ever access the web gui, not to mention actually control the LEDs. No reset is occuring.

### To Reproduce Bug

OPNSense router? Unsure if this makes a difference.
Happening on latest 0.14.0-b4, on ALL devices except a single esp I had running. I've tried over 11 different ESP devkits, esp32, esp32c3, and esp32s2 to no avail.

### Expected Behavior

It stays friggin connected to the wifi network!

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b4

### Which microcontroller/board are you seeing the problem on?

ESP32, ESP32-S2, ESP32-C3

### Relevant log/trace output

```shell
Connected! IP address: 192.168.1.105
IMPROV�
Init STA interfaces
mDNS started
Disconnected!
Access point disabled (init).
Connecting to Shhhh...
Last reconnect too old.
Access point disabled (init).
Connecting to Shhhh...
Loop took 82ms.
Usermods took 0ms.
Strip took 1ms.
Local time: 00:12
---DEBUG INFO---
Runtime: 720370
Unix time: 720,370
Free heap: 195124
Wifi state: 6
State time: 630322
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 2670
Loop time[ms]: 8/83
UM time[ms]: 0/1
Strip time[ms]: 0/8
Segments: 1 -> 68B
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
Buffer: 3*30=90B

Connected! IP address: 192.168.1.105
IMPROV�
Init STA interfaces
mDNS started
JSON buffer locked. (17)
JSON buffer size: 7786 for request: 0
JSON content length: 4167
Loop took 33ms.
Usermods took 0ms.
Strip took 1ms.
JSON buffer released. (17)
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tache, 2023-09-13:
I was having this issue with my USB-C devices. Using the USB-C plug, they were unusable. I then cut a USB cable and connected wires directly to VIn and GND and I am not having no issues.

blazoncek, 2023-09-13:
Might be a noise on RX/TX pins forcing WLED into Improv config mode.

closed 2024-04-23 by github-actions[bot]

dvbit, 2024-10-22:
@Apprisco i am experiencing exactly the same issues with similar network devices. Opnsense se and unifi