wled/WLED #3364: ESP8266 not compiling with rotary encoder and display usermods

srg74, 2023-09-08:
### What happened?

Getting error while compiling ESP8266 binary with 2 usermods encoder and 4 line display

### To Reproduce Bug

Here is env I build with:
`[env:esp8266_display_rotary]
board = esp12e
platform = espressif8266@3.2.0
upload_speed = 460800
platform_packages = ${common.platform_packages}
board_build.ldscript = ${common.ldscript_4m1m}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266}
  -D WLED_RELEASE_NAME=esp8266_display_rotary
  -D WLED_DISABLE_BLYNK
  -D WLED_DISABLE_CRONIXIE
  -D WLED_DISABLE_HUESYNC
  -D WLED_DISABLE_LOXONE
  -D WLED_DISABLE_MQTT
  -D WLED_DISABLE_ALEXA
  -D USE_ALT_DISPlAY ; new versions of USERMOD_FOUR_LINE_DISPLAY and USERMOD_ROTARY_ENCODER_UI
  -D USERMOD_FOUR_LINE_DISPLAY
  -D RLYPIN=-1
  -D BTNPIN=-1
  -D LEDPIN=2
  -D IRPIN=-1
  -D USERMOD_ROTARY_ENCODER_UI
  -UWLED_USE_MY_CONFIG
lib_deps = ${esp8266.lib_deps}
  olikraus/U8g2 @ ^2.35.5`

### Expected Behavior

Expected to be compiling as have been before.

### Install Method

Self-Compiled

### What version of WLED?

v0.14.0-b4 latest build

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pio/build/esp8266_display_rotary/firmware.elf section `.text1' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp8266_display_rotary/firmware.elf] Error 1
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-09-09:
Use the flag: `-DPIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48`
The code is simply to large to fit into default IRAM.

srg74, 2023-09-09:
Tested, works fine! Thanks for help!

closed 2023-09-09 by srg74