wled/WLED #3367: No Conmection after Update 0.14.0-b5

mc-gl89, 2023-09-10:
### What happened?

After i updated 2 of my 3 WLed over Homeassistant they are offline. The Lighta shows yelow ans Green Lights. Poweroff dosent helps. Reset over Buttons dosent help. No AP Availabel.

### To Reproduce Bug

Update WLed over Homeassistant Updatefunction

### Expected Behavior

Updatein and reconmect Wled over Wlan

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b4

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WarC0zes, 2023-09-11:
Hi,
I updated my ATOM Matrix from 0.14.0-b4 to 0.14.0-b5 by HomeAssistant. The update was done but I got an error message at the end. The ATOM Matrix works fine in 0.14.0-b5.

LOGS:
```
Logger: homeassistant.components.websocket_api.http.connection
Source: components/wled/helpers.py:37
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 08:35:59 (1 occurrences)
Last logged: 08:35:59

[547499661504] Invalid response from WLED API
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/aiohttp/client_reqrep.py", line 899, in start
    message, payload = await protocol.read()  # type: ignore[union-attr]
                       ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/streams.py", line 616, in read
    await self._waiter
  File "/usr/local/lib/python3.11/site-packages/aiohttp/client_proto.py", line 213, in data_received
    messages, upgraded, tail = self._parser.feed_data(data)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "aiohttp/_http_parser.pyx", line 557, in aiohttp._http_parser.HttpParser.feed_data
aiohttp.http_exceptions.BadHttpMessage: 400, message:
  Invalid character in chunk size:

    b'181 \r'
         ^

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/wled/wled.py", line 699, in upgrade
    await self.session.post(url, data=form)
  File "/usr/local/lib/python3.11/site-packages/aiohttp/client.py", line 560, in _request
    await resp.start(conn)
  File "/usr/local/lib/python3.11/site-packages/aiohttp/client_reqrep.py", line 901, in start
    raise ClientResponseError(
aiohttp.client_exceptions.ClientResponseError: 400, message="Invalid character in chunk size:\n\n  b'181 \\r'\n       ^", url=URL('http://192.168.1.33:80/update')

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/wled/helpers.py", line 28, in handler
    await func(self, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/components/wled/update.py", line 89, in async_install
    await self.coordinator.wled.upgrade(version=version)
  File "/usr/local/lib/python3.11/site-packages/wled/wled.py", line 711, in upgrade
    raise WLEDUpgradeError(msg) from exception
wled.exceptions.WLEDUpgradeError: Could not download requested WLED version '0.14.0-b5' from https://github.com/Aircoookie/WLED/releases/download/v0.14.0-b5/WLED_0.14.0-b5_ESP32.bin

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 227, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1969, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2006, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 870, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 942, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 157, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 428, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/wled/helpers.py", line 37, in handler
    raise HomeAssistantError("Invalid response from WLED API") from error
homeassistant.exceptions.HomeAssistantError: Invalid response from WLED API

```

softhack007, 2023-09-11:
```
Could not download requested WLED version '0.14.0-b5' from
https://github.com/Aircoookie/WLED/releases/download/v0.14.0-b5/WLED_0.14.0-b5_ESP32.bin
```

🤔 URL used is correct ... looks like the error happened when HA tried to download the file from github? 
If yes, this would be a problem within HA, as they fail to parse the downloaded file ... Just an educated guess.

softhack007, 2023-09-11:
@Aircoookie another thought - do you know if HA uses IMPROV ? 
I remember that you updated the improv code for -b4, maybe HA cannot parse the additional information that is provided by WLED?

Aircoookie, 2023-09-11:
@softhack007 Improv is serial-only as implemented in WLED, so the HA integration would not be using it.

I faintly recall there being a bug with the semver parser in HA that lead to a problem in the integration with an alpha or beta of 0.13.0, not sure which one it was specifically...

softhack007, 2023-09-15:
> I faintly recall there being a bug with the semver parser in HA that lead to a problem in the integration with an alpha or beta of 0.13.0, not sure which one it was specifically...

Now that you mention, yes I can recall an unpleasant discussion with a HA core developer some time ago. The problem was that HA could not parse the "-b5" part of our version, leading to all kinds of strange effects. The response we got was something like "well if I have to chose between fixing it for your case or leaving it as-is, then obviously I'll chose to leave it as-is" (in short 😉).

https://github.com/Aircoookie/WLED/issues/2616

blue-hound, 2023-10-05:
I had this problem too. I had to turn off 'Lock wireless (OTA) software update' and I think clear the passphrase. Then Home Assistant was able to trigger the update. Is there some way to tell Home Assistant what the OTA password is?

blazoncek, 2024-01-06:
Related to #3641 and others.

blazoncek, 2024-01-09:
Closing as corrupt JSON has been fixed in 0.14.1-b3.
If the issue persist in 0.14.1-b3 or later, please open a new issue.

closed 2024-01-09 by blazoncek