wled/WLED #3370: Ripple Effect broken on matrix

poelzi, 2023-09-12:
### What happened?

When the LEDs are configured as matrix, the ripple effect behaves flickery

### To Reproduce Bug

1. Configure strip as matrix
2. Use one segment with ripple effect

### Expected Behavior

1d & 2d strip should look similar

### Install Method

Self-Compiled

### What version of WLED?

ESP32

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-09-12:
Ripple effect has been upgraded to work on 2D as ripples on the water.
It is intended that way.

If you want the same behaviour set up segment as 1D.

blazoncek, 2023-09-12:
https://github.com/Aircoookie/WLED/assets/7722392/49f6d9e2-f16a-4e28-aed6-e199e72b733d

softhack007, 2023-09-15:
> the ripple effect behaves flickery

Ripple is a very "fast" effect, so quick changes in pixel brightness could appear as flickering, especially on a matrix setup.

If you have several segments, each segment has its own brightness slider. I've noticed that changing segment brightness (aka opacity) is sometimes stronger compared to using the global brightness slider. 

So one thing you could try: Set segment brightness to max, and use the main brightness slider for adjusting your lights. Make sure you have "Use global LED buffer" enabled in LED settings, and check if the "max current" for brightness limiter is adequate for your fixture.

blazoncek, 2023-09-15:
We may tweak 2D ripples to add trail to lessen the apparent flickering.
What do you think @softhack007 ?

softhack007, 2023-09-15:
> We may tweak 2D ripples to add trail to lessen the apparent flickering. What do you think @softhack007 ?

I can check the effect on my larger matrix tomorrow, but yeah why not add a small fade_out() as option, similar to what we did for the rollingBalls effect.

blazoncek, 2023-09-16:
@poelzi can you try https://github.com/Aircoookie/WLED/commit/87cc3fd714fe7c5a3dd806f8dbd529493f171d9d and see if it behaves any better?

blazoncek, 2023-09-28:
Fix available in b6.

closed 2023-09-28 by blazoncek