wled/WLED #3371: Expand 1D FX setting should not change automatically

freakintoddles2, 2023-09-12:
**Is your feature request related to a problem? Please describe.**
When using 2d panels, some 1d effects will override/change the current 'Expand 1D FX' setting in segments to Arc or Bar. This is not expected behavior since this setting is part of the segment. Just like I don't expect the start/stop x/y pixels, groups, spacing or other segment options to change without my input, I also don't expect this to change unless I change it. 

The problem this causes is that I have to do a lot of extra work to create presets for everything with this setting saved as part of the preset. I can't just rely on changing effects alone and expecting it will not change the current Expand 1D FX setting.

**Describe the solution you'd like**
Decouple this Expand 1D FX setting from the effects that currently force it to change. Do not change the Expand 1D FX setting based on effects changing. 

**Describe alternatives you've considered**
Only reasonable alternative currently is spending a lot of time saving presets with the expand 1d FX setting set to Pixels, but that is time consuming and not friendly for people who just want to try out effects from the main list.

**Additional context**
n/a

Thank you for your consideration here!

Thank you for your ideas for making WLED better!

blazoncek, 2023-09-12:
What you expect is your personal preference and the designer of the effect may have had other preference.
As some effects may look bad with certain mappings (or other parameters) a designer choose to make certain defaults. These are applied only when clicking/selecting effect in UI. If you use API or create a preset those are not applied.
This is the intended behaviour.

That being said, I do agree that this may be confusing for novice user.

As a workaround for the functionality you are after (trying every effect) you can create a single preset that will cycle through all effects on each press without changing any other parameter.
The simplest would be: `{"seg":{"fx":"~"}}` using JSON API or `FX=~` using HTTP API.

As a discussion we can ask @ewoudwijma who designed that behaviour what would be his take on this problem.

freakintoddles2, 2023-09-12:
Thanks, that is a helpful workaround, I'll give it a shot.  I do understand that the effect designer has a perspective, but it creates a UX inconsistency within segments for the reasons I already mentioned. 

In particular, the additional trouble it causes as it works today is that once it overrides the setting to Arc/Bar, that override stays in place until it's changed back even on effects that do not have this override. So suddenly the setting is changed and now applying to all future selected effects. I don't consider that an issue only affecting novice users, it's unexpected and inconsistent behaviour.

dosipod, 2023-09-12:
The workaround we did for a long time was to add "m12":0   to api  , 
 {"seg":[{"id":0,"fx":"r","m12":0}]}

Does not seem that bad , and yes once we use ledmaps in some cases we do not actually want  Pixels  but bars , might be the case that it is a personal preference  but defaults should be standardized else they are no longer default but at random .

dosipod, 2023-09-12:
Just noticed the discussion on discord  so yes it will work if the defaults is unchecked

blazoncek, 2023-09-13:
Yes, I forgot about that option in UI settings. I think @freakintoddles2 can close the issue.

closed 2023-09-13 by freakintoddles2