wled/WLED #3372: Home Assistant - API communication issue

jeroen85, 2023-09-14:
### What happened?

WLED functions correctly, via HA and web UI.
After some time, it stops functioning in HA.
Continues to function correctly via web UI.

A reboot of WLED solves the issue for some time (minutes).

### To Reproduce Bug

Start using WLED via HA.
Wait a bit.
It stops working in HA.

### Expected Behavior

Continue to work :-)

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0-b5

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
2023-09-14 19:37:41.672 DEBUG (MainThread) [homeassistant.components.wled] Finished fetching wled data in 0.083 seconds (success: True)
2023-09-14 19:37:51.671 DEBUG (MainThread) [homeassistant.components.wled] Finished fetching wled data in 0.082 seconds (success: True)
2023-09-14 19:38:01.686 DEBUG (MainThread) [homeassistant.components.wled] Finished fetching wled data in 0.098 seconds (success: True)
2023-09-14 19:38:14.087 ERROR (MainThread) [homeassistant.components.wled] Error fetching wled data: Invalid response from API: Error occurred while communicating with WLED device at 192.168.2.6
2023-09-14 19:38:14.089 DEBUG (MainThread) [homeassistant.components.wled] Finished fetching wled data in 2.500 seconds (success: False)
2023-09-14 19:38:25.245 DEBUG (MainThread) [homeassistant.components.wled] Finished fetching wled data in 1.656 seconds (success: False)
2023-09-14 19:38:36.306 DEBUG (MainThread) [homeassistant.components.wled] Finished fetching wled data in 1.717 seconds (success: False)
2023-09-14 19:38:48.240 DEBUG (MainThread) [homeassistant.components.wled] Finished fetching wled data in 2.651 seconds (success: False)
2023-09-14 19:39:00.200 DEBUG (MainThread) [homeassistant.components.wled] Finished fetching wled data in 2.611 seconds (success: False)
2023-09-14 19:39:11.032 DEBUG (MainThread) [homeassistant.components.wled] Finished fetching wled data in 1.443 seconds (success: False)
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-09-14:
Not a WLED bug *per se*. You will need to talk to HA team.

jeroen85, 2023-09-14:
Found an issue with IP addresses. Both WLED and a phone had the same IP-address. 
Fixed this issue, looks like this caused the API problems (which makes sense)

closed 2023-09-14 by jeroen85