wled/WLED #3374: Effect Behavior Changes in b5

BRUXXUS, 2023-09-16:
### What happened?

I have been using the beta releases as they come out without issue, but when updating to b5, my default preset, Meteor no longer functions the same.
Previously, using the *Colors Only and *Color Gradient palettes would allow me to set a background color, then the other two would be the leading edge of the meteor, which would gradient into the trail of the meteor.

With b5, the effect now spreads all colors in the palette across the whole segment, and the meteor changes color as it travels. 

This behavior also changes Meteor Smooth, Multi Comet, and other effects with a leading edge and trail. 

### To Reproduce Bug

Load up Meteor, Meteor Smooth, or Multi Comet effect, then use any of the 2-3 color custom palettes. 

### Expected Behavior

When using any of the palettes which allow for custom 1-3 colors, one should be a background, another should be leading edge of the effect, and another be the trailing effect of the moving portion. 

### Install Method

Binary from WLED.me

### What version of WLED?

v0.14.0-b5

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

[Discord Link to video demonstrating operation pre b5](https://discord.com/channels/473448917040758787/779395228624617512/1152305976846864556)

[Discord Link to video demonstrating behavior after b5 - same preset settings](https://discord.com/channels/473448917040758787/779395228624617512/1152306063060770903)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-09-16:
Color (slot) 2 is usually background color, not 1. That's consistent in many effects. But...

If you use any palette other than Default, color slots are ignored and only palette is used (a very few effects may ignore or partially ignore that).
Using palette starting with * is still palette though it is generated from color slots.

If you are using ESP32 I would recommend you enable double buffering in LED settings and see if that makes color rendering any better.

It is also true that some effects were intentionally modified for 0.14. A Github blame (or commit history) can show if that holds true for Meteor.

blazoncek, 2023-09-16:
Update: Change happened a year ago while trying to bring behaviour of Meteor palette use closer to behaviour of other effects.
https://github.com/Aircoookie/WLED/commit/6ba3e25d33337262b7fe9ce0cfd278d224801a2d

What is your *desired* outcome was in fact an inconsistency (and perhaps a bug) in behaviour regarding color slot use in Meteor effect. v0.14 introduced *slider control* which allows hiding certain UI elements depending on other selections and also tried to bring consistency to how effects behave.
Meteor effect only used palette (effect C++ code) and did not use any of the slots. If you selected Default palette its output was not anything like "meteor".
[image]
(code from before April 2022)

I will be adding an option to set background color as is your *desired* behaviour while keeping current palette use. This will unfortunately require that you change preset but you will be able to achieve same end result.

blazoncek, 2023-09-28:
Fix available in b6.

closed 2023-09-28 by blazoncek