wled/WLED #3375: White Balance correction setting breaks the matrix effect, and others

freakintoddles2, 2023-09-16:
### What happened?

If the 'White Balance correction' setting is enabled, it breaks the matrix effect.
It only lights up the top row of pixels and everything below is off. 
Funky plank may also be affected, at least blazoncek made it sound like it did. 

### To Reproduce Bug

Enable White Balance correction, select matrix effect 

### Expected Behavior

It works

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.04b

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-09-17:
Thanks for the report. I'll look I'd we can do something for matrix; the problem is that the effect expects exactly one color value to start a new stream. If you have "global buffer" off, or you use color temperature correction, then the value returned by getPixelColor() is not the same as the one previously set.

We might be able to do something for matrix, however our getPixelColor can't undo color temperature corrections. Maybe this part of the problem needs to be addressed in 0.15.x.

blazoncek, 2023-09-17:
@softhack007 the problem is `if (pix == spawnColor)` which is never `true` as `pix = SEGMENT.getPixelColorXY(col, row)` which will return *whitebalance corrected* color.

The same is in several SR effects which relied on `leds[]` which was immutable after setting.

blazoncek, 2023-09-17:
I am thinking of moving white balance correction into `BusDigital::show()` which will immediately fix such issues when using double buffering.
But non-double-buffered output will suffer.

softhack007, 2023-09-17:
@blazoncek yes that sounds reasonable, but not sure what we could do for people who don't use global LED buffer to save RAM?

Just for sake of complete mess, Matrix also does not work together with scrolling text overlay - matrix "drops" end at the first letter.

I'll think of a solution for Matrix, PR will come in the next days.

blazoncek, 2023-09-17:
One thing is certain: If effect relies on previous pixel values we'd need to re-introduce `setUpLeds()`.

softhack007, 2023-09-17:
> the problem is `if (pix == spawnColor)` which is never `true` as `pix = SEGMENT.getPixelColorXY(col, row)` which will return _whitebalance corrected_ color.

@blazoncek true,  just took a closer look, and my conclusion is the whole thing is over-complicated. The double loop can be simplified, because new spawns only happen in the first row, everything else is just "move down". A lossless buffer for the first row is enough. I'll try that tomorrow.

blazoncek, 2023-09-18:
I wouldn't say it is over complicated or that 1 line is enough. The trail does fade.
Perhaps @ewoudwijma could say more as he is credited in the source.

When we have overlapping segments each segment should have it's own buffer (`setUpLeds()`) to allow unmodified effects to run properly otherwise effect will need to take into account that pixel may not stay the same between frames or even between `setPixelColor()` and `getPixelColor()`. White balance will ruin the latter for sure. This may be something for @Aircoookie as he was the one to implement white balance correction and knows its workings.

softhack007, 2023-09-18:
I've committed a fix that should make Matrix more robust against color corrections, however its not perfect. For example with color correction enabled and a very "red" color set-point, all trails will slowly change their colors into the red.

I think that's all we can do in 0.14.0 - as @blazoncek says, a full solution would mean a re-write of white balancing, and we would need to bring back `setUpLeds()` to allow effects to access unmodified colors that are not affected by gaps or overlays. 

BTW, tested the "Funky Plank" effect and did not see any similar problems. Was maybe just be a rumor...
If you find other effects that misbehave, please open a new bug report.

blazoncek, 2023-09-18:
Funky plank has issues with gaps (or ledmap with missing pixels).

softhack007, 2023-09-18:
> Funky plank has issues with gaps (or ledmap with missing pixels).

Makes sense, I think most effects that include some scrolling (= moving previous LEDs left/right/up/down) will stop showing anything once they hit a gap or un-mapped pixel. Especially audio-reactive effects are affected, but there are also others that maybe won't work with some kind of "black holes" in between.

My point was: to solve these issues, we probably need to change segment class implementation, so I'd try to avoid this refactoring work at least for 0.14.0 and 0.14.1.

blazoncek, 2023-09-19:
> My point was: to solve these issues, we probably need to change segment class implementation, so I'd try to avoid this refactoring work at least for 0.14.0 and 0.14.1.
> 
> 

God forbid! 😀 Segment class is almost perfect.

What needs to be done is move ledmaps and gaps into Bus class (instead of WS2812FX class).
When that is done, use gaps or color manipulation when sending to underlying NeoPixelBus and the problem is solved.

At the expense of double buffering or even tripple buffering for overlapping segments.

softhack007, 2023-10-01:
Closing, as the original problem with Matrix effect is solved with 0.14.0-b6.
I think that gap and ledmap problems should be discussed in another place.

closed 2023-10-01 by softhack007