wled/WLED #3378: Remap addressable led chips like ws2811 as separate LED channels

ThisAccount7777, 2023-09-19:
**Is your feature request related to a problem? Please describe.**
Problem: Each led has to be controlled by a separate WS2811 Chip so 24 LEDS == 24 Chips used and PCB space.

**Describe the solution you'd like**
The ability to remap all the channels on an led controller like the WS2811 as 3 separate channels that can only be controlled from 0-255. 

> Suppose you would have 3 LEDS connected to a WS2811 and you want to control each RGB channel as if it was a separate led with its own index the same way you would with WS2811 white.

blazoncek, 2023-09-19:
That is implemented a white addressable.

closed 2023-09-19 by blazoncek

ThisAccount7777, 2023-09-19:
Ok

ThisAccount7777, 2023-09-19:
> That is implemented as white addressable.

I couldn't find it

blazoncek, 2023-09-19:
[image]
[image]
[image]

ThisAccount7777, 2023-09-19:
[image]

What I'm saying is the ability to split the 3 led channels of the WS2811 ic into 3 separate addressable channels.

blazoncek, 2023-09-19:
That's how it is implemented. Each LED channel acts as a single white led with brightness values 0-255.
It cannot be done otherwise.
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.

snenyl, 2025-03-26:
This is not implemented correctly. 

Current implementation of "WS2811 White" is:

LED 1 is controller 1 color R and controller 2 color R and controller 3 color R
LED 2 is controller 1 color G and controller 2 color G and controller 3 color G
LED 3 is controller 1 color B and controller 2 color B and controller 3 color B
LED 4 is controller 4 color R and controller 5 color R and controller 6 color R
LED 5 is controller 4 color G and controller 5 color G and controller 6 color G
LED 6 is controller 4 color B and controller 5 color B and controller 6 color B

But is should be:

LED 1 is controller 1 color R
LED 2 is controller 1 color G
LED 3 is controller 1 color B
LED 4 is controller 2 color R
LED 5 is controller 2 color G
LED 6 is controller 2 color B

Here is what happens to my LED strip when starting each LED from 1 to 6:

[image]
[image]
[image]
[image]
[image]
[image]

blazoncek, 2025-03-27:
There are plenty of different implementations of "single addressable white" LED strips. WLED currently supports one that was 1st available (and may be most common).
Unfortunately I do not own any of these and the current implementation is from @Aircoookie if he can lend help.