wled/WLED #3379: Indexing the channels on WS2811 as if there were 3 separate leds.

ThisAccount7777, 2023-09-20:
**Is your feature request related to a problem? Please describe.**
Hardware limitations - read context for more info.

**Describe the solution you'd like**
Index each channel on the addressable led chips like WS2811/WS2814 as a separate led/index
eg.
[ic: 1, channel: r, index: 1], [ic: 1, channel: g, index: 2], [ic: 1, channel: b, index: 3], [ic: 2, channel: r, index: 4], [ic: 2, channel: g, index: 5]

**Describe alternatives you've considered**
74HC595 and Arduino (limited control over leds) or one WS2811 per led

**Additional context**
[image]
Suppose you are are trying to control 6 LEDS but there isn't enough room on the PCB to accommodate all the chips. 
Note that this is an EXAMPLE circuit so it wouldn't work. The final version will have 21 leds.

blazoncek, 2023-09-20:
Implemented as WS2811 White (type 19) in LED settings. If this does not satisfy your requirements you will need to re-implement yourself.
Duplicate of #3378, closing.

closed 2023-09-20 by blazoncek

blazoncek, 2023-09-20:
Using "WS2811 White" LED type controls each channel ow WS2811 chip as a separate white (or any color) single pixel (index) LED.
This is exactly what you are describing.

ThisAccount7777, 2023-09-21:
Control each channel on any addressable led driver ic as a single addressable LED*

softhack007, 2023-09-21:
> Control each channel on any addressable led driver ic as a single addressable LED*

@ThisAccount7777 the only difference between your last statement and the conclusion from @blazoncek is "WS281x chip" vs "any addressable led IC". For me this sound like the same thing, please clarify if I misunderstood.

I've seen the complaints in discord about "MODs keep closing my request because they do not understand". So you need to help us understand.

I you are looking for something different than what we already have, there is some homework for you :
* try "WS2811 White" on your hardware
* summarize how the behaviour differs from what you are looking for.

Then we have a basis for discussion.

softhack007, 2023-09-21:
Maybe for additional  clarification: 
WLED communicates with the ws281x IC, but it cannot "see" or control anything behind the IC. So it's possible to "abuse" R/G/B color and treat then as three different "white" items. But in case that each ws2811 has several LEDs attached to each color output (like most ws2811 control 3 leds at once) this is something we cannot change.

ThisAccount7777, 2023-09-22:
I might have to try again on a physical setup with WS2811 white mode.