wled/WLED #3381: Compile ERROR

Francesco-Rugg, 2023-09-22:
### What happened?

When I try to compile wled with usermod Always when I open a .h file from a usermod I got a bunch of error.
I'm trying to compile wled with display and rotary encoder (I know there's the srg repository that can it be used but i want to compile a bin that already have my option and learn)
I don't know if the errors founds by VScode in .h files are related with the terminal error when i'm try to build but is strange that even if I follow all the steps in the readme files of usermods I always got an error in the build process.

the error is for example usermode v2 four line display `#include <U8x8lib.h>`  error that says:
#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (C:\Users\rugge\Desktop\WLED\usermods\usermod_v2_four_line_display\usermod_v2_four_line_display.h).C/C++(1696)
cannot open source file "U8x8lib.h"C/C++(1696)

[image]
[image]
[image]

For the usermod I proceed with this steps:

First setup the environment ecc ecc in the platformio override like this
` [platformio]
default_envs = d1_mini

[env:d1_mini]
board = d1_mini
platform = ${common.platform_wled_default}
platform_packages = ${common.platform_packages}
upload_speed = 921600
board_build.ldscript = ${common.ldscript_4m1m}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266}
    -D usermod_v2_four_line_display
    -D usermod_v2_rotary_encoder_ui
    -D USE_ALT_DISPlAY

lib_deps = ${esp8266.lib_deps}
    U8g2
    Wire
monitor_filters = esp8266_exception_decoder`

then copy the .h file of the corresponding usermod in wled00
and always got an error

### To Reproduce Bug

try to compile usermod

### Expected Behavior

create a build for wemos d1 mini

### Install Method

Self-Compiled

### What version of WLED?

WLED beta release 0.14.0-b5

### Which microcontroller/board are you seeing the problem on?

ESP8266, Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-09-22:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
Closing.

closed 2023-09-22 by blazoncek

softhack007, 2023-09-22:
@Francesco-Rugg to explain a bit what you see:
* the "redline" marks are normal, they are not from compiling the source code. The VisualStudio Code editor has problems to follow all the source code dependencies. The marks are no problem for building the firmware. There is nothing planned to change this, as a major restructuring of our source code would be needed just to make the VSCode editor happy. So its tiny added value, but lots of work to do that.
* overwriting an existing build environment in your `platformIO_override.ini` is not good style (i.e. `[env:d1_mini]`). It will be better if you give your build environment a new name, like `[env:my_own_d1_mini]`.

In future, please use github tickets only when reporting WLED software bugs, and for suggesting new features for WLED.
For everything else, we have [discord](https://discord.gg/KuqP7NE) where many helpful community members are regularly reading new posts in the public channels.

Francesco-Rugg, 2023-09-22:
> @Francesco-Rugg to explain a bit what you see:
> * the "redline" marks are normal, they are not from compiling the source code. The VisualStudio Code editor has problems to follow all the source code [...]

Ok, thank you, I'm totally new to this word and these are my first try to compile something, I was erroneously thinking that the problem was related to wled files. 
Sorry and thank you again