wled/WLED #3382: SyntaxError and sometimes a reset when changing presets in 2D.

MOVE1904, 2023-09-23:
### What happened?

I made an illuminated sign for the back under my jersey for my favorite club.

[image]

There are 540 LEDs arranged irregularly on the sign. I configured WLED so that I use this in a 2D Matrix.
When I choose a presets I often got a syntax error or a reset of the ESP32, even with the simplest configuration.

https://github.com/Aircoookie/WLED/assets/43117741/fbf7413b-02f8-4a3f-a8ca-c2a9140e20a4

### To Reproduce Bug

Use the attached cfg and preset file.

[preset_cfg.zip](https://github.com/Aircoookie/WLED/files/12706610/preset_cfg.zip)

Select the “Start” preset several times.

### Expected Behavior

I expected that there would be no reset or error when changing the preset.

### Install Method

Self-Compiled

### What version of WLED?

WLED v0.14.0-b5 "Hoshi" (Build 2309100)

### Which microcontroller/board are you seeing the problem on?

ESP32, ESP32-S3

### Relevant log/trace output

```shell
---- Opened the serial port COM4 ----
WS FileRead: /index.htm
Not-Found HTTP call:
URI: /skin.css
WS FileRead: /skin.css
JSON buffer locked. (17)
JSON buffer size: 7890 for request: 6
JSON content length: 5285
JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 4801 for request: 8
JSON content length: 2184
JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 11798 for request: 3
JSON content length: 6665
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
WS FileRead: /presets.json
WS client connected.
JSON buffer locked. (12)
JSON buffer size: 10967 for WS request (6665).
heap 181664
Sending WS data to a single client.
JSON buffer released. (12)
JSON buffer locked. (14)
JSON buffer released. (14)
JSON buffer locked. (17)
JSON buffer size: 11799 for request: 3
JSON content length: 6673
JSON buffer released. (17)
JSON buffer locked. (12)
JSON buffer size: 10968 for WS request (6673).
heap 154840
Sending WS data to multiple clients.
JSON buffer released. (12)
Loop took 45ms.
Usermods took 0ms.
Strip took 7ms.
Local time: 13:19
Local time: 13:20
JSON buffer locked. (14)
JSON buffer released. (14)
JSON buffer locked. (17)
JSON buffer size: 11799 for request: 3
JSON content length: 6673
JSON buffer released. (17)
JSON buffer locked. (12)
JSON buffer size: 10968 for WS request (6673).
heap 173916
Sending WS data to multiple clients.
JSON buffer released. (12)
Loop took 68ms.
Usermods took 0ms.
Strip took 8ms.
---DEBUG INFO---
Runtime: 30000
Unix time: 1695475204,000
Free heap: 176456
Wifi state: 3
State time: 0
NTP last sync: 999000000
Client IP: XXX.XXX.XXX.XXX
Loops/sec: 1727
Loop time[ms]: 0/721
UM time[ms]: 0/9
Strip time[ms]: 0/160
Segments: 16 -> 1088B
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*1024=2048B
Buffer: 3*1024=3072B
---- Closed the serial port COM4 ----
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MOVE1904, 2023-09-23:
[preset_cfg_2.zip](https://github.com/Aircoookie/WLED/files/12706848/preset_cfg_2.zip)

[ledmap1.zip](https://github.com/Aircoookie/WLED/files/12706847/ledmap1.zip)

Here is a bigger configuration with more frequent errors. Switch between preset 23 (PixelsFire) and preset 14 (Start).

blazoncek, 2023-09-23:
For JSON errors you need browser's developer tools and see console log.

blazoncek, 2023-09-23:
Also preset file in the movie does not correspond to attached presets.

MOVE1904, 2023-09-23:
> Also preset file in the movie does not correspond to attached presets.

Did You use the preset file from my first post?

[preset_cfg.zip](https://github.com/Aircoookie/WLED/files/12706932/preset_cfg.zip)

MOVE1904, 2023-09-23:
> For JSON errors you need browser's developer tools and see console log.

like this?

[image]

blazoncek, 2023-09-23:
Now check each GET JSON response to see content. One will have error.
[image]

MOVE1904, 2023-09-23:
No JSON. Only 1 CSS maybe?

[image]

[image]

MOVE1904, 2023-09-23:
O.K. I got it. CSS is enable custom CSS in User Settings. So I diabled it.

So where can I find the GET JSON with an error?

blazoncek, 2023-09-24:
Any file with JSON type. I.e. `effects`, `fxdata`, etc.

MOVE1904, 2023-09-24:
This looks strange

[image]

blazoncek, 2023-09-24:
Yes. That's corrupt JSON.
Why that happens is still a mystery.

blazoncek, 2023-09-24:
If you are not using ESP8266 then check if you flashed your ESP with non-standard flash or CPU speed (lowering the values to standard specs).

If you are using ESP8266 I've seen such behaviour after we upgraded core to 3.1.2 but is transient in nature. Some responses may be corrupted. I am guessing some out-of-order new/free behaviour of AsyncWebServer or its components.

MOVE1904, 2023-09-24:
I'm flashing the ESP32 with the standard option of VS Code. Upload port is set to "auto".
I try it with 6 different ESP32-WROOM-32 an 2 different ESP32-S3-WROOM-S1.
Always the same errors. Most of them with position 5533.

Here ESP32-S3
[image]

And here ESP32-WROOM-32
[image]

blazoncek, 2023-09-24:
Looks like the issue is connected to the number of segments or the size of segment data.

blazoncek, 2023-09-24:
Additional info:

This is UI only problem as the HTTP response contains corrupt data (HTTP headers inserted into HTTP response payload).

The issue (most likely) arises from the following code in `json.cpp` `serveJson()`:
```c++
  AsyncJsonResponse *response = new AsyncJsonResponse(&doc, subJson==JSON_PATH_FXDATA || subJson==JSON_PATH_EFFECTS); // will clear and convert JsonDocument into JsonArray if necessary
  JsonVariant lDoc = response->getRoot();
...
  response->setLength();
  request->send(response);
```

Looks very much similar to what I've observed on ESP8266. Reducing the number of segments to 14 alleviates the problem. WS responses are not affected.

MOVE1904, 2023-09-24:
> Looks like the issue is connected to the number of segments or the size of segment data.

I deleted 3 segments. Now I can not reproduce the error. It doesn't matter which 3 segments I delete, so I think the maximum number of segments when using ledmap is 13 !?!

blazoncek, 2023-09-24:
It is not ledmap related. At least I think so.

MOVE1904, 2023-09-24:
> It is not ledmap related. At least I think so.

You're right. I mean 2D matrix. My preset does not use a ledmap file.

blazoncek, 2023-09-28:
Similar to #3312

softhack007, 2023-10-03:
@blazoncek this is the first "smoking gun" with a full debug build - looks like stack overflow. I'll continue to investigate.

```
***ERROR*** A stack overflow in task loopTask has been detected.
abort() was called at PC 0x4008a890 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008a4d4:0x3ffafe00 0x4008a879:0x3ffafe20 0x4008a890:0x3ffafe40 0x4008ce17:0x3ffafe60 0x4008ebc0:0x3ffafe80 0x4008eb76:0x3ffc0ee8 |<-CORRUPTED
  #0  0x4008a4d4:0x3ffafe00 in invoke_abort at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:699
  #1  0x4008a879:0x3ffafe20 in abort at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:699
  #2  0x4008a890:0x3ffafe40 in vApplicationStackOverflowHook at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:699
  #3  0x4008ce17:0x3ffafe60 in vTaskSwitchContext at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
  #4  0x4008ebc0:0x3ffafe80 in _frxt_dispatch at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:406
  #5  0x4008eb76:0x3ffc0ee8 in _frxt_int_exit at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:206

Rebooting...
```

blazoncek, 2023-10-03:
Debug and stack use are enemies.
Funny there isn't any WLED code involved. Makes me wonder where that happens.

softhack007, 2023-10-03:
> Debug and stack use are enemies. Funny there isn't any WLED code involved. Makes me wonder where that happens.

Seems you are right - it only happens with `build_type = debug`; crazy workaround is to double arduino stacksize with `-DARDUINO_LOOP_STACK_SIZE=16896`.

So back to zero. At least I can reproduce the UI JSON error.

blazoncek, 2023-10-03:
> At least I can reproduce the UI JSON error.

I can do that, but not consistently.

softhack007, 2023-10-03:
@blazoncek can you cross-check something?

I've changed this line in ws.cpp
https://github.com/Aircoookie/WLED/blob/26e766ee1976491fbce698b24ce3a65d3850a855/wled00/ws.cpp#L111

to 
```
  size_t len = measureJson(doc)+1;
```

With this change I do not get the UI error any more. But not sure if this is a real fix, or its just hiding the symptoms. What do you think?

<br />

the docs for measureJson say: 
> Computes the length of the minified JSON document that [serializeJson()](https://arduinojson.org/v6/api/json/serializejson/) produces, excluding the null-terminator.

So if we want to serialise to a string, one more char is needed for the terminating '\0'.

MOVE1904, 2023-10-03:
Added this change but still got errors on an ESP32-S3.

[image]

[image]

[image]

softhack007, 2023-10-04:
Update: the change I tried in ws.cpp does not help, but causes other errors.

My current best guess: the same same JSON is sent out twice - maybe even in parallel - and confuses the UI.

async_tcp: serveJson() responds to /json/si
```
   JSON buffer locked. (17)
   JSON buffer size: 11958 for request: 3
   JSON content length: 6813
```
looptask: shortly afterwards, sendDataWs(); is called from updateInterfaces()
```
JSON buffer locked. (12)
JSON buffer size: 11127 for WS request (6813).
Sending WS data to multiple clients.
```

blazoncek, 2023-10-04:
This (double notification) should not be an issue.
Never was.

softhack007, 2023-10-05:
@MOVE1904 please check if this preliminary fix solves the problem for you.
* https://github.com/Aircoookie/WLED/pull/3427 

its a workaround, so I expect that the problem may not disappear completely, but it should happen less often.

MOVE1904, 2023-10-05:
> @MOVE1904 please check if this preliminary fix solves the problem for you.
> 
> * [workaround for #3382 #3427](https://github.com/Aircoookie/WLED/pull/3427)
> 
> its a workaround, so I expect that the problem may not disappear completely, but it should happen less often.

First quick test: No more syntax error, but still (maybe more) "Connection to light failed!".

MOVE1904, 2023-10-05:
First test on an ESP32-WROOM-32 and no Preset with more than 13 segments

Second test on an ESP32-S3 and some presets with 16 segments:
Still got syntax errors, "Failed to fetch" and "Connection to light failed!" with and without reset.

blazoncek, 2023-10-05:
Please build without websockets `-D WLED_DISABLE_WEBSOCKETS`

MOVE1904, 2023-10-05:
Still the same, sorry. No presets with more than 13 segment.
I can't say that it is harder or not to generate the errors.

[image]
[image]
[image]

blazoncek, 2023-10-05:
I can no longer reproduce with @softhack007 fix and WS still enabled.
[image]

softhack007, 2023-10-05:
I'm running out of ideas. Seems that we should check through the ~75 open PRs for [asyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer/pulls), maybe someone already invented a fix that makes the library more thread-safe... and if we find something, let's ask @Aircoookie to update his branch for WLED.

It means we will keep this bug open in 0.14.0, as the test results are still "inconclusive" and the risk to accidentally break some other functions is just too high.

MOVE1904, 2023-10-05:
For me it's okay if the error can't be completely fixed. Since I don't use more than 13 segments, I've found ways to avoid the error.

There are much more importent things to do. Let's go ...

softhack007, 2023-10-06:
This patch for asyncWebserver _might_ help, however seems  it does not improve behaviour on 8266
* https://github.com/me-no-dev/ESPAsyncWebServer/pull/887

blazoncek, 2023-10-06:
Would it be possible that these two (#3295) are related?

blazoncek, 2023-12-31:
I invite you to test `0_15` branch and possibly `suspend` branch, both employ some additional stuff to try to mitigate the cause.
If I do not hear back within a month (either confirming that issue is gone or persists) I am closing as unresolved.

MOVE1904, 2023-12-31:
I still get the error "Connection to light failed!

0_15 branch:
[image]

suspend branch:
[image]

Additionally I got the error "Error 8: Effect RAM depleted!"

0_15 branch:
[image]

But overall the errors have become fewer.

blazoncek, 2023-12-31:
The only explanation (suggested by your last screenshot) is that you run out of heap. Reduce your segment complexity or use ESP with PSRAM.
It may also help if you show browser's console log.

MOVE1904, 2023-12-31:
Here is a simpler configuration of my Christmas lights:

[WLED_cfg_preset.zip](https://github.com/Aircoookie/WLED/files/13801158/WLED_cfg_preset.zip)

[image]

blazoncek, 2023-12-31:
One more thing.
Reviewing configuration (again), I am baffled with `total` LED count of 1024 while the sum of all lengths yields only 540. How is that so?
And if you are using matrix (as evident from config) and custom ledmap (`ledmap1.json`) where is your default ledmap?
Do you switch ledmaps often?

blazoncek, 2023-12-31:
And what is the last response from WLED (network tab, click on the last message).

MOVE1904, 2023-12-31:
> One more thing. Reviewing configuration (again), I am baffled with `total` LED count of 1024 while the sum of all lengths yields only 540. How is that so? And if you are using matrix (as evident [...]

In order to not use so many segments and still be able to display a graphic, I created an LED-MAP. Since I only use 2 colors, I first have all white and then all blue LEDs in the LED-MAP.

> And what is the last response from WLED (network tab, click on the last message).

Sorry, I'm not that good with Chrome. I hope that's correct. It is with the simpler configuration.

[image]

MOVE1904, 2023-12-31:
In general, I have to say that I see these errors more and more often in my projects. It doesn't matter whether I use an ESP32 or D1Mini and whether I use LED MAPs or not.
It doesn't matter whether I use a ready to use image or whether I compiled it myself.
These errors even appear in the Android app.

blazoncek, 2023-12-31:
> In order to not use so many segments and still be able to display a graphic, I created an LED-MAP. Since I only use 2 colors, I first have all white and then all blue LEDs in the LED-MAP.

That does not explain `total` having 1024. There is no way you could achieve that using UI and the configuration you posted.

I give up. You'll need to debug your issue on your own and if you indeed find a bug please report back.

MOVE1904, 2023-12-31:
I give up, too. 
Maybe it's another type of the Fritz!Box error.
I just think it's a shame that you only pay attention to the strange arrangement of my LED lighting, but don't address the fact that the error now appears even in the simplest configurations.
I only implemented the idea with the LED-Map to reduce the crashes. At least the paylists are stable with this solution.

closed 2023-12-31 by MOVE1904