wled/WLED #3385: esp32-c3-mini-1 can't boot

chs72, 2023-09-23:
### What happened?

esp32-c3-mini-1 can't boot

### To Reproduce Bug

esp32-c3-mini-1 can't boot

### Expected Behavior

booting

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.14.0-b5_ESP32-C3

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Relevant log/trace output

```shell
Using 'COM8' as serial port.
Connecting....
Detecting chip type... ESP32-C3
Connecting...

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-C3 (revision 3)
 - Number of Cores: 1
 - Max CPU Frequency: 80MHz
 - Has Bluetooth: NO
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: NO
 - MAC Address: **:**:**:**:**:**
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dout
 - Flash Frequency: 80MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 17.2s
Flash will be erased from 0x00000000 to 0x00002fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x0017cfff...
Flash params set to 0x032f
Compressed 12112 bytes to 8764...
Writing at 0x00000000... (100 %)
Wrote 12112 bytes (8764 compressed) at 0x00000000 in 0.4 seconds (effective 231.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 402.2 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 518.5 kbit/s)...
Hash of data verified.
Compressed 1490992 bytes to 918347...
Writing at 0x00010000... (1 %)
Writing at 0x00017cea... (3 %)
Writing at 0x0001bbd1... (5 %)
Writing at 0x0001fbcc... (7 %)
Writing at 0x00023bee... (8 %)
Writing at 0x00027c09... (10 %)
Writing at 0x0002bc0e... (12 %)
Writing at 0x0002ff52... (14 %)
Writing at 0x0003aa4e... (15 %)
Writing at 0x000444ef... (17 %)
Writing at 0x000511ca... (19 %)
Writing at 0x0005c1ee... (21 %)
Writing at 0x00066fdf... (22 %)
Writing at 0x0006e05c... (24 %)
Writing at 0x00074197... (26 %)
Writing at 0x0007b211... (28 %)
Writing at 0x00081c9b... (29 %)
Writing at 0x00088694... (31 %)
Writing at 0x0008ec74... (33 %)
Writing at 0x00094caa... (35 %)
Writing at 0x0009b57c... (36 %)
Writing at 0x000a17bb... (38 %)
Writing at 0x000a7cf1... (40 %)
Writing at 0x000ae34b... (42 %)
Writing at 0x000b4faf... (43 %)
Writing at 0x000bb1e7... (45 %)
Writing at 0x000c1fd6... (47 %)
Writing at 0x000c7de6... (49 %)
Writing at 0x000ce538... (50 %)
Writing at 0x000d4ca1... (52 %)
Writing at 0x000dc31c... (54 %)
Writing at 0x000e2a79... (56 %)
Writing at 0x000e8c8e... (57 %)
Writing at 0x000eeae4... (59 %)
Writing at 0x000f46f2... (61 %)
Writing at 0x000fa516... (63 %)
Writing at 0x00100710... (64 %)
Writing at 0x0010645f... (66 %)
Writing at 0x0010c48f... (68 %)
Writing at 0x00112403... (70 %)
Writing at 0x00118004... (71 %)
Writing at 0x0011de4a... (73 %)
Writing at 0x0012354b... (75 %)
Writing at 0x00129555... (77 %)
Writing at 0x0012fd41... (78 %)
Writing at 0x0013585c... (80 %)
Writing at 0x0013ba3f... (82 %)
Writing at 0x00141d92... (84 %)
Writing at 0x0014828d... (85 %)
Writing at 0x0014e864... (87 %)
Writing at 0x00154aef... (89 %)
Writing at 0x0015a8b0... (91 %)
Writing at 0x00160ed0... (92 %)
Writing at 0x00168912... (94 %)
Writing at 0x0016f68f... (96 %)
Writing at 0x001752c4... (98 %)
Writing at 0x0017b995... (100 %)
Wrote 1490992 bytes (918347 compressed) at 0x00010000 in 26.0 seconds (effective 458.4 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[02:37:10]ESP-ROM:esp32c3-api1-20210207
[02:37:10]Build:Feb  7 2021
[02:37:10]rst:0xf (BROWNOUT_RST),boot:0x4 (DOWNLOAD(USB/UART0/1))
[02:37:10]waiting for download
[02:37:15]ESP-ROM:esp32c3-api1-20210207
[02:37:15]Build:Feb  7 2021
[02:37:15]rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
[02:37:15]SPIWP:0xee
[02:37:15]mode:DOUT, clock div:1
[02:37:15]load:0x3fcd6100,len:0x38c
[02:37:15]load:0x403ce000,len:0x6ac
[02:37:15]load:0x403d0000,len:0x24e4
[02:37:15]SHA-256 comparison failed:
[02:37:15]Calculated: 536dfe3b000041806b8028e5cb5125cd46736c37326abcaa489747c6c6fe4906
[02:37:15]Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
[02:37:15]Attempting to boot anyway...
[02:37:15]entry 0x403ce000
[02:37:16]E (302) esp_core_dump_flash: No core dump partition found!
[02:37:16]E (302) esp_core_dump_flash: No core dump partition found!
[02:37:16]./components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
[02:37:16]E (315) esp_littlefs: mount failed,  (-84)
[02:37:16]E (319) esp_littlefs: Failed to initialize LittleFS
[02:37:22]ESP-ROM:ESPESP-ROM:esp32c3-api1-20210207
[02:37:22]Build:Feb  7 2021
[02:37:22]rst:0xf (BROWNOUT_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
[02:37:22]SPIWP:0xee
[02:37:22]mode:DOUT, clock div:1
[02:37:22]load:0x3fcd6100,len:0x38c
[02:37:22]load:0x403ce000,len:0x6ac
[02:37:22]load:0x403d0000,len:0x24e4
[02:37:22]SHA-256 comparison failed:
[02:37:22]Calculated: 536dfe3b000041806b8028e5cb5125cd46736c37326abcaa489747c6c6fe4906
[02:37:22]Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
[02:37:22]Attempting to boot anyway...
[02:37:22]entry 0x403ce000
[02:37:22]E (303) esp_core_dump_flash: No core dump partition found!
[02:37:22]E (303) esp_core_dump_flash: No core dump partition found!
[02:37:25]ESP-ROM:esp32c3-api1-20210207
[02:37:25]Build:Feb  7 2021
[02:37:25]rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
[02:37:25]SPIWP:0xee
[02:37:25]mode:DOUT, clock div:1
[02:37:25]load:0x3fcd6100,len:0x38c
[02:37:25]load:0x403ce000,len:0x6ac
[02:37:25]load:0x403d0000,len:0x24e4
[02:37:25]SHA-256 comparison failed:
[02:37:25]Calculated: 536dfe3b000041806b8028e5cb5125cd46736c37326abcaa489747c6c6fe4906
[02:37:25]Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
[02:37:25]Attempting to boot anyway...
[02:37:25]entry 0x403ce000
[02:37:25]E (302) esp_core_dump_flash: No core dump partition found!
[02:37:25]E (302) esp_core_dump_flash: No core dump partition found!
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-09-23:
Hi,

It seems you have a hardware (wiring) problem. Your boards is  doing "brownout resets", meaning the power supply voltage has dropped too low. 

This is the brownout:
`02:37:22]Build:Feb 7 2021 [02:37:22]rst:0xf (`**`BROWNOUT_RST`**`),boot:0xc (SPI_FAST_FLASH_BOOT`

This part is normal:
```
02:37:15]entry 0x403ce000 [02:37:16]E (302) esp_core_dump_flash: No core dump partition found!
[02:37:16]E (302) esp_core_dump_flash: No core dump partition found!
[02:37:16]./components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}`
```

blazoncek, 2023-09-27:
I just flashed ESP32-C3 DEVKITM-1 without issues using web installer.
The problem must be with flasher you are using or your hardware environment. Please use [web flasher](https://install.wled.me) or use Espressif `esptool`.

closed 2023-10-06 by chs72