wled/WLED #3387: Changing presets corrupts presets.json

bscholer, 2023-09-26:
### What happened?

Whenever I change my presets on one specific device, and refresh the page, I get an error saying that the presets.json file is corrupt and cannot be loaded.

When I inspect the file on the edit page, I can find the syntax issue quickly. It seems like when I save a preset, it duplicates it without a key in the presets array. As you can't have keyless objects in JSON, this results in a corrupted file.

### To Reproduce Bug

1. Download [presets.zip](https://github.com/Aircoookie/WLED/files/12730450/presets.zip), unzip, and upload `presets.json` file to any WLED device, running 0.13.X or 0.14.X (maybe older too).
2. Navigate to home page, and see that everything is working as expected.
3. Make a change to an existing preset (overwrite with state)
4. Refresh the page after saving, and see that the presets.json file is now corrupted.

### Expected Behavior

I should be able to save changes to existing presets without corrupting the file.
Better yet, WLED should validate that the new JSON is valid before writing over the old one.

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.X, 0.14.X

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

This only seems to happen with this presets.json file, and I'm not really sure why. The file I linked in the "To reproduce bug" section is valid, and nothing seems to be wrong with it. Everything falls apart when you edit a preset though.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-09-26:
A known issue. Root cause still unknown.
0.14 should be less affected though.

bscholer, 2023-09-26:
Are there other GitHub issues about this? I'm guessing some amount of debugging has been done on it?

blazoncek, 2023-09-26:
Yes, plenty. I no longer have the energy for it.

bscholer, 2023-09-27:
#2793, closing

closed 2023-09-27 by bscholer

blazoncek, 2025-12-29:
Anyone stumbling on this: Presets on ESP32 get corrupted (99.999%) on save by incompatible flash mode used when flashing ESP for the first time.
To resolve the issue *you will need to flash new bootloader using DIO or DOUT mode*.