wled/WLED #3388: Pwm Output Value Is Not Zero (0-255). For Rgbw

seyfettinkazanir, 2023-09-26:
### What happened?

When I select PWM RGBW from the LED features, it does not reach zero when I want to turn off the LEDs completely using the invert option.

### To Reproduce Bug

WLED 13.3 and 14.6 installation process was done. PWM RGBW option was selected from LED features. Pin numbers were chosen as 12, 4, 14, 15. Since transistors were used, the invert option was activated. Observation was made with an oscilloscope. When we wanted to turn off the LEDs completely, the brightness value was set to zero, but PWM did not reach zero.

### Expected Behavior

The LEDs were expected to turn off completely.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 13.3 ve WLED 14.6

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

[image]

The oscilloscope image was taken when PWM was set to zero. (invert option is active.)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-09-26:
WLED uses hardware LEDC driver for PWM. Check for bugs in Espressif code.

seyfettinkazanir, 2023-09-26:
[image]

The last value for brightness settings appears to be 1. Could it be related to this error?

blazoncek, 2023-09-27:
That is normal behaviour. If you turn WLED off (using UI button) the brightness will be 0.

closed 2023-12-03 by blazoncek