wled/WLED #3390: esp02.bin.gz is missing from 0.14.0-b6

jflecool2, 2023-09-27:
### What happened?

esp02.bin.gz is missing from 0.14.0-b6.
Its required to do manual OTA upgrade, as the .bin file is too big.
Thanks

### To Reproduce Bug

go in releases page

### Expected Behavior

see a file called "WLED_0.14.0-b6_ESP02.bin.gz"

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.14.0-b6

### Which microcontroller/board are you seeing the problem on?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jflecool2, 2023-09-27:
PS: compressed to .bin.gz myself, renamed the bin within to "esp8266_2m.bin", did manual OTA upgrade and it worked. So yes, .gz is required for manual OTA upgrade in some cases.

softhack007, 2023-10-01:
File added, thanks for the hint 👍 

@Aircoookie: We need a way to add bin.gz files to artifacts of GH action builds. Maybe replace all .bin for 8266 with bin.gz. Would be nice if you could look into this, once your time allows.

closed 2023-10-01 by softhack007