wled/WLED #3391: Disable sound processing while lights are off

protyposis, 2023-09-27:
When the AudioReactive Sync Mode is set to `Send`, sound processing always runs, i.e., even when the lights are powered off. It would be great if sound processing could be turned on/off with the lights, like it can be done manually with the toggle button on the Web UI's info page.

In my use case, I only use audio reactive light effects when all WLED lights are turned on, so an always running sound process unnecessarily floods the network with a UDP broadcast, and maybe it even unnecessarily uses power?

(WLED v0.14.0-b6)

softhack007, 2023-09-30:
Hi, I understand you use case, however there are other use cases that would not work any more with your proposal.

Your proposal make another use case impossible however : running audioreactive as a "sound server" with no lights connected. Clients with light will connect or disconnect as they like, so the "server" should always send data.

Some additional technical points:
* you can use normal "sync" feature to make other "clients" use the master brightness (so they all go out together)
* if you remove the "enabled" checkmark in usermod settings, then also UDP sound sync should stop sending.
* audioreactive is a usermod, and currently there is no way to hook into the "light out" switch in the main UI. 
* power consumption should not be a big concern, especially on "classic esp32" and on esp32-s3. Audioreactive itself does not cause more than 10-15% of CPU load on these boards.
* udp sound sync is 44bytes each 20ms, so 2200bytes per second. We use multicast, so the number of packets does not increase with more clients connected.

protyposis, 2023-09-30:
Thank you for the detailed feedback.

Anyway, if this was ever implemented, it should be optional of course to keep the sound server use case. Another future workaround could be to expose usermod settings via the API and toggle audio reactive mode with an automation.

closed 2023-09-30 by protyposis

blazoncek, 2023-09-30:
@protyposis you can always disable audio processing from UI.
There is also an API for that if you want to use it programmatically.

protyposis, 2023-09-30:
> There is also an API for that if you want to use it programmatically.

That would be great! I tried the JSON API, but it seemed to be read-only (`curl -X POST "http://<wled-ip>/json/state" -d '{"AudioReactive":{"on":false}}' -H "Content-Type: application/json"`).

blazoncek, 2023-09-30:
use just /json
and make sure your quotes are correct ones

protyposis, 2023-09-30:
Thanks, I figured it out :) While the property is called `on` in the response,

```
// /json/state
{
  "on": true,
  // ...
  "AudioReactive": {
    "on": true
  },
  // ...
}
```
it needs to be called `enabled` in the request: `curl -X POST "http://<wled-ip>/json/state" -d '{"AudioReactive":{"enabled":false}}' -H "Content-Type: application/json"`

(See https://github.com/Aircoookie/WLED/blob/5eadbe7ecd21fedc60d1fec7066c382014e9b968/usermods/audioreactive/audio_reactive.h#L1597 vs. 
https://github.com/Aircoookie/WLED/blob/5eadbe7ecd21fedc60d1fec7066c382014e9b968/usermods/audioreactive/audio_reactive.h#L1611-L1614)