wled/WLED #3392: Fehler bei Verbindungsversuch mit Alexa

c206hr, 2023-09-27:
### What happened?

Hallo,
zuerst einmal möchte ich sagen das ich ihre Arbeit super finde und hoffe das es noch weiter entwickelt wird.
jetzt zu meinem Problem,
ich habe heute den ganzen Tag versucht mein WLED in Alexa einzurichten jedoch wurde dieses nicht gefunden. Ich verwende als Board einen ESP32-WROOM-32 worauf die Applikation WLED super läuft nur bei dem versuch dieses mit Alexa zu verbinden bin ich gescheitert. Ich habe einige Lösungsansätze versucht jedoch immer ohne Erfolg. Ich muss jedoch dazu sagen das ich es nicht geschafft habe mit meinem wissen einen Echtzeit log auszulesen da ich mich nicht unbedingt so gut in dem ganzen Thema auskenne. 
Ich bin jetzt doch recht ratlos und würde mich über eine Antwort sehr freuen, vielleicht habe ich ja doch etwas übersehen. 

Mit freundlichen Grüßen 
 

### To Reproduce Bug

Ich habe WLED ganz normal über die Installations Seite Installiert und eingerichtet. Also WLAN sowie die Aktivierung der Alexa Funktion. Nach dem Neustart des ESP habe ich dann auf dem Router UPNP aktiviert und versucht über die Alexa App den ESP zu finden jedoch ohne erfolg

### Expected Behavior

Habe eigentlich gedacht das ich das aufgrund der letzten Einträge und Fehler die schon beantwortet wurden auf die reihe bekommen werde jedoch war dies nicht der Fall. Habe eigentlich mit einer schnellen und einfachen Einrichtung in Alexa gerechnet 

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0-b6 und 0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-09-29:
Can you write your problem report in english, please? Most members of the WLED core team do not speak german.

c206hr, 2023-09-29:
Okay, yes no problem.
Excuse me, but there are mistakes in my English because I speak English
very poorly

What happened?
Hello,
First of all I would like to say that I think your work is great and I hope
that it will be further developed.
Now to my problem,
I tried to set up my WLED in Alexa all day today, but it couldn't be found.
I use an ESP32-WROOM-32 as a board in connection with the WLED super
application, but I failed when I tried to connect it to Alexa. I've tried a
few solutions, but always without success. However, I have to say that I
wasn't able to read a real-time log with my knowledge because I don't
necessarily know that much about the whole topic.
I'm really at a loss now and would really appreciate an answer, maybe I've
overlooked something.

Best regards

To reproduce the error
I installed and set up WLED as normal via the installation page. Also WiFi
and the activation of the Alexa function. After restarting the ESP, I
activated UPNP on the router and tried to find the ESP using the Alexa app,
but without success

Expected behavior
I actually thought that because of the last entries and errors that have
already been answered in the series, this was not the case. I actually
expected it to be quick and easy to set up in Alexa

Installation method
Binary from WLED.me

Which WLED version?
WLED 0.14.0-b6 and 0.13.3

Which microcontroller/board are you seeing the problem on?
ESP32

GrayMatter, 2024-01-15:
I have a similar issue.  I have a D1 mini board running a string of 60 WS2811 LED's. It's only function is Alexa integration to use voice commands to turn the LEDs on and off, change colors, etc. It had been running for over a year flawlessly without any firmware upgrades then one day Alexa said the device was unresponsive. I rebooted and no change so I performed a fresh default install of the firmware to 0.14.0-b2 which was the latest at the time and reconfigured it from scratch. It worked as it had before until about a week ago and now Alexa can no longer see it again. I can still see it and control it through its local web interface over WiFi so I know it hasn't dropped off the network and it is still functional.  I haven't started debugging the issues yet but just doing an OTA update to the current version hasn't helped. I'm wondering if Amazon made changes to their API that WLED uses?

robson1311, 2024-05-08:
same problem as c206hr.
Is there a solution now?

GrayMatter, 2024-05-30:
> same problem as c206hr. Is there a solution now?

Just checked release notes for the new builds and no mention of Alexa so I'm guessing it's still potentially a problem. This doesn't seem to be a feature that many use so may be a while.  If I have time this weekend, I'll take a look at it again. For now I just have mine running 24x7 with a solid colour but have the annoying flash bug that has been mentioned a few times before.

closed 2024-10-06 by github-actions[bot]