wled/WLED #3398: ESP reboots every x minutes when connected to my network

efiten, 2023-09-29:
### What happened?

ESP is rebooting about every 5 minutes when connected to my Unifi (WiFi) network.

### To Reproduce Bug

Difficult one. Checked with several other users, they don't have this.
Ubiquity Unifi network (USG3 & UAP-AC-Pro & U6-Pro), with latest firmware.
Separate IoT network, only 2.4Ghz active, very basic.

Several D1 mini running on the same network, with ESPHome, no issues at all.

Tested with 2 different boards : 
- https://quinled.info/quinled-esp32/
- https://www.tindie.com/products/serg74/esp32-wroom-usb-c-d1-mini32-form-factor-board/ 

Both experiencing EXACTLY the same behaviour.

When WLED is in AP mode, and laptop connected to WLED-AP, there is no problem, no reboots. As soon as I change the WiFi network to my home network, the ESP reboots after around 5 minutes.
I tried a separate WiFi network with an old DLINK AP (DAP-1350), connected to my LAN, same behaviour, reboot every 5 minutes.
Setup a Hotspot with my GSM, connected my laptop and the ESP, no problem at all, no reboots.

flashed the 2 ESP's with ESPHome, connected them to the same IoT network, no reboots. So it has to be some kind of interaction between my network (not even WiFi, because it happens with a separate AP on my network also), and the WLED image.

Installed the DEBUG version, and connected over USB to capture the log.

### Expected Behavior

ESP with WLED should remain online and not reboot.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0-b6 2309241

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Relevant log/trace output

```shell
Runtime: 270127
Unix time: 270,127
Free heap: 178808
Wifi state: 3
State time: 0
NTP last sync: 999000000
Client IP: 192.168.2.6
Loops/sec: 1808
Loop time[ms]: 4/3
UM time[ms]: 0/1
Strip time[ms]: 0/2
Segments: 1 -> 68B
Modes: 4*187=748B
 Data: 4*187=748B
 Map: 2*0=0B
 Buffer: 3*300=900B
Local time: 00:05
 ---DEBUG INFO---
Runtime: 300143
Unix time: 300,143
Free heap: 104256
Wifi state: 3
State time: 0
NTP last sync: 999000000
Client IP: 192.168.2.6
Loops/sec: 1613
Loop time[ms]: 6/3
UM time[ms]: 0/1
Strip time[ms]: 0/2
Segments: 1 -> 68B
Modes: 4*187=748B
 Data: 4*187=748B
 Map: 2*0=0B
 Buffer: 3*300=900B
abort() was called at PC 0x4014504a on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008982c:0x3ffb1d90 0x40089b89:0x3ffb1db0 0x4014504a:0x3ffb1dd0 0x401ab9c8:0x3ffb1df0 0x401ab98d:0x3ffb1e10 0x40124033:0x3ffb1e30 0x4011161d:0x3ffb1e70 0x4011433c:0x3ffb1f50 0x40114d5a:0x3ffb1f90 0x4013363d:0x3ffb1fb0 0x4008b5d2:0x3ffb1fd0

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

---WLED 0.14.0-b6 2309241 INIT---
esp32 v3.3.6-16-gcc5440f6a2
arduino-esp32 v1.0.x

CPU:   ESP32-D0WDQ5 rev.3, 2 core(s), 240MHz.
FLASH: 4MB, Mode 3 (DOUT), speed 40MHz.
heap 292060
PIN ALLOC: Pin 1 successfully allocated by 0x89 (137)
Registering usermods ...
heap 292060
Reading config
Reading settings from /wsec.json...
JSON buffer locked. (3)
JSON buffer released. (3)
JSON buffer locked. (1)
Reading settings from /cfg.json...
Removing all.
PIN ALLOC: Pin 16 successfully allocated by 0x82 (130)
Successfully inited strip 0 (len 300) with type 22 and pins 16,186 (itype 21)
             PIN ALLOC: Pin 0 successfully allocated by 0x85 (133)
Starting usermod config.
JSON buffer released. (1)
Initializing strip
Loading custom palettes
Loading custom ledmaps
heap 286200
Usermods setup
heap 286200
initServer
heap 282168
lastReconnectAttempt == 0
Access point disabled (init).
Connecting to Fiten-Reekmans-WLED...
Loop took 135ms.
Usermods took 0ms.
 Strip took 3ms.
 Loop took 717ms.
 Usermods took 0ms.
 Strip took 2ms.

Connected! IP address: 192.168.2.6
Init STA interfaces
mDNS started
WS client connected.
JSON buffer locked. (12)
JSON buffer size: 2164 for WS request (1185).
heap 185608
Sending WS data to a single client.
JSON buffer released. (12)
---DEBUG INFO---
Runtime: 30000
Unix time: 30,000
Free heap: 200904
Wifi state: 3
State time: 0
NTP last sync: 999000000
Client IP: 192.168.2.6
Loops/sec: 2518
Loop time[ms]: 0/717
UM time[ms]: 0/1
Strip time[ms]: 0/3
Segments: 1 -> 68B
 Modes: 4*187=748B
 Data: 4*187=748B
 Map: 2*0=0B
 Buffer: 3*300=900B
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-09-29:
I know this may be frustrating but is not a WLED bug as WiFi stack is provided by Espressif framework.
As you already figured out other WiFi equipment will not produce reboots so it is definitely your Unifi setup causing the issues.

I own Ubiquiti Unifi with several APs (not v6 though) and never had any issues with that.

Please try modifying Unifi parameters. Start by disabling everything and then gradually enable features. I would suspect that meshing my be the main reason but I would not limit search to it.

efiten, 2023-09-29:
That's why I added an old (DLINK DAP-1350) as AccessPoint  wired to my network, setup a separate WiFi network with ONLY B&G, no other options (not possible in this old firmware). And even on this WiFi network, the reboot occurs. So I think it's not WiFi related, but network related, some strange packet that triggered the ESP to reboot ?
BUT, with different firmware (ESPHome, I think it uses the same framework), there is no problem.
I have Other ESP's running (some ESPhome, and : https://github.com/dl9rdz/rdz_ttgo_sonde ), and those are running for months now, on the same network. So there has to be some difference, not ?

efiten, 2023-09-29:
[image]
[image]

blazoncek, 2023-09-29:
Get a Wireshark and monitor network traffic. You should also get a debug build and capture serial output from WLED.
Post both here.

efiten, 2023-09-29:
The log posted in the issue is from the debug build, I already did this on request from 'Annihlator' on Discord.
Halfway the log, the restart is happening, with debug info before and after.

blazoncek, 2023-09-29:
`abort()` is not called by WLED code.
Can you load exception decoder to see where in call stack it is called from? You may need to compile with PIO for that.

efiten, 2023-09-29:
OK, and there you lost me :)
I'll come back when I figured this out.

blazoncek, 2023-09-29:
You will need to add `monitor_filters = esp32_exception_decoder` in your PIO build environment.

I.e.
```
[env:esp32custom]
board = esp32dev
platform = ${esp32.platform}
platform_packages = ${esp32.platform_packages}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp32}
  -D WLED_DEBUG
lib_deps = ${esp32.lib_deps}
monitor_filters = esp32_exception_decoder
monitor_port = /dev/cu.SLAB_USBtoUART # or your serial port
board_build.partitions = ${esp32.default_partitions}
```
And then hit "Upload and monitor".

efiten, 2023-09-29:
I was doing some more tests, created a separate VLAN on that WiFi network, and disabled Multicast DNS for that VLAN.
[image]

And, for now, the ESP is running for more than 30mins.
The only disadvantage is 'no auto discovery' for HA running on a separate VLAN. So for now, I'll wait and do some more testing

blazoncek, 2023-09-29:
HA is known to cause WLED resets. Please post your Wireshark captures.

softhack007, 2023-09-29:
> I was doing some more tests, created a separate VLAN on that WiFi network, and disabled Multicast DNS for that VLAN.

> And, for now, the ESP is running for more than 30mins. The only disadvantage is 'no auto discovery' for HA running on a separate VLAN. So for now, I'll wait and do some more testing

Interesting 👍 
The overall problem scenario reminds me of this problem report - the only known remedy is to self-compiled WLED with a new framework version. Unfortunately this framework is not well tested, and it adds a whopping 300kB to the firmware size. But if might be worth a try [[env:esp32dev_V4_dio80]](https://github.com/Aircoookie/WLED/blob/f442aad9620b93e4ae15bc9160e73f5d25eb6d15/platformio.ini#L418C6-L418C23).

* https://github.com/Aircoookie/WLED/issues/2932#issuecomment-1609913421

closed 2023-12-03 by blazoncek