wled/WLED #3399: Gamma Correction for Custom Palettes

BaconWithThat, 2023-09-29:
**Is your feature request related to a problem? Please describe.**
I am trying to implement custom palettes and the output doesn't match my expectation.  After some experimentation, the output matches what I see with gamma correction turned off.  After some chat on discord, it was confirmed that custom palettes are expected to be gamma corrected.

**Describe the solution you'd like**
I think the most straightforward solution would be to do gamma correction for custom palettes.

**Describe alternatives you've considered**
If I knew how to do the gamma correction manually, I could do that and save the corrected values as a custom palette.

**Additional context**
Thanks to Blazoncek in the discord for helping me.

blazoncek, 2023-09-30:
Just a clarification for further consideration:
Existing FastLED palettes are already *gamma corrected* (at least kind of) but custom palettes are not.
Custom palette editor (available since 0.14.0-b2) does not know of any gamma correction and uploads palette as entered in UI. Also, it is possible to enter palette using /edit maintenance page.
So, custom palettes will remain gamma unadjusted in the file. We can adjust gamma when loading such palette from file into RAM.
Unfortunately this will have performance impact as loading can happen thousands of times per second.

BaconWithThat, 2023-10-01:
Could we get instructions on how to do gamma correction as an article on the wiki?  Then users w/ custom palettes could put the gamma corrected colors into their custom palettes.  I'm unsure what the downside would be aside from it looking wonky on the computer screen.

blazoncek, 2023-10-01:
https://en.m.wikipedia.org/wiki/Gamma_correction

BaconWithThat, 2023-10-01:
I ended up using the values in the built in [Gamma correction table ](https://github.com/Aircoookie/WLED/blob/332be7edd69853f70aebf92b83a4f0f275bcfa96/wled00/colors.cpp#L305) to manually transpose my custom palette and it looks great.

blazoncek, 2023-10-01:
I've implemented gamma correction for custom palettes.
Will be available in 0.14.1

closed 2024-01-04 by blazoncek